RPi-Distro/pi-gen #642: dphys-swapfile erases /etc/fstab

booo, 2022-11-01:
We encountered a case where dhpys-swapfile erases the /etc/fstab file. This happens as soon as the file system does not have enough space for the swap file and dphys-swapfile starts to create smaller swap files. dphys-swapfile erases the /etc/fstab only if we remove power from the board in the right moment. This happened quite a lot with our system because we were debugging other ports of the system and cut power frequently. I guess some write/move operations on the /etc/fstab are not atomic and therefore if luck has it we end up with an empty or deleted /etc/fstab.

XECDesign, 2022-11-01:
Yeah, that doesn't sound great. Might be worth reporting to https://bugs.debian.org as well.

I don't think it's related to any changes we've made to the package, so it's probably not going to be the highest priority, but if you'd like to send a patch, I can add it sooner.

XECDesign, 2024-01-02:
Closing, since it's an upstream issue.

closed 2024-01-02 by XECDesign