RPi-Distro/pi-gen #643: setcap calls dropped inside chroot

benfrancis, 2022-11-03:
The [WebThings Gateway](https://github.com/WebThingsIO/gateway/) open source project provides a [Raspberry Pi image](https://webthings.io/gateway/), [built](https://github.com/WebThingsIO/gateway/blob/master/image/build.sh) using pi-gen's `build-docker.sh`.

Part of that build process [uses](https://github.com/WebThingsIO/gateway/blob/master/image/stage3/01-install-deps/01-run-chroot.sh#L6) `setcap` inside `stage3/01-install-deps/01-run-chroot.sh` to give Python and Node.js scripts access to Bluetooth hardware, which is essential for the functioning of this IoT gateway software.

```
setcap cap_net_raw+eip $(eval readlink -f $(which python3))
setcap cap_net_raw+eip $(eval readlink -f $(which node))
```

This used to work, but in current builds these lines appear to be [failing](https://github.com/WebThingsIO/gateway/issues/2985) inside chroot because pi-gen [uses](https://github.com/RPi-Distro/pi-gen/blob/master/scripts/common#L93) the `--drop=cap_setfcap` flag in its `on_chroot()` function.

This causes our builds to fail but seems a bit tricky to fix without making an upstream change to pi-gen.

Do you know the reason that `--drop=cap_setfcap` is used, and does this seem like the likely cause? Can you recommend either a fix in pi-gen or a workaround in WebThings Gateway's build system?

This is a critical build error blocking quite an urgent bugfix release for us.

Many thanks for this very useful project!

XECDesign, 2022-11-03:
> This causes our builds to fail but seems a bit tricky to fix without making an upstream change to pi-gen.

I can see such a config parameter being useful.

> Do you know the reason that `--drop=cap_setfcap` is used, and does this seem like the likely cause? Can you recommend either a fix in pi-gen or a workaround in WebThings Gateway's build system?

It's used because not all filesystems support capabilities. For example, netbooting with an NFS root, wouldn't work well.

For now, if it's an urgent release, you can just remove the `--drop=cap_setfcap` option.

benfrancis, 2022-11-07:
@XECDesign Thanks very much for your response.

> For now, if it's an urgent release, you can just remove the `--drop=cap_setfcap` option.

I don't suppose you have any advice on how to achieve that?

Our build.sh calls pi-gen's build-docker.sh, which it clones from GitHub. Is there any way for us to override this command from our [stage3](https://github.com/WebThingsIO/gateway/tree/master/image/stage3) directory, or would I have to write a script which modifies pi-gen's scripts/common script after cloning it?

XECDesign, 2022-11-07:
That one approach. Another one might be to clone your own fork instead.

I've put adding a better approach to my todo list and will update the issue when that's sorted.

closed 2022-11-08 by XECDesign

benfrancis, 2022-11-11:
Thanks @XECDesign, this is great!

Do you have a policy about backporting fixes to previous releases?

I could do with the SETFCAP feature on the buster branch. I tried backporting your patch to the buster branch in a new [fork](https://github.com/WebThingsIO/pi-gen/), but for some reason the feature didn't seem to work for me.

For now we're just [manually](https://github.com/WebThingsIO/pi-gen/commit/a865d161833237df508f1bbeec59fb7c4aa836d1) removing the `--drop=cap_setfcap` argument from the `setarch` commands in a fork.

XECDesign, 2022-11-14:
> Do you have a policy about backporting fixes to previous releases?

No problem with backporting simple low risk changes like this.

I've cherry-picked the commit and it seems to work okay on this end.

commit c3083ecd (2022-11-08): Add parameter to keep cap_setfcap

Fixes #643
commit adbc45f3 (2022-11-14): Add parameter to keep cap_setfcap

Fixes #643