RPi-Distro/pi-gen #644: build-docker.sh and build.sh - grabs i386 debian image

juddy, 2022-11-05:
root@debian:/home/wjk/Code/pi-gen# ./build-docker.sh 
Sending build context to Docker daemon  906.2kB
Step 1/6 : ARG BASE_IMAGE=debian:bullseye
Step 2/6 : FROM ${BASE_IMAGE}
bullseye: Pulling from **i386**/debian
8caf426a2703: Pull complete

juddy, 2022-11-05:
### Steps to reproduce:

    git clone --depth 1 https://github.com/RPI-Distro/pi-gen.git

Create a config file and run

    sudo ./build.sh

### Workaround:

    git clone --depth 3 https://github.com/RPI-Distro/pi-gen.git

XECDesign, 2022-11-07:
Why is this an issue and how does changing the clone depth fix it?

alexander-sony, 2022-12-05:
Is it possible to build arm64/v8 images on a x86_64 host? The `case "$(uname -m)" in` in build-docker.sh must be there for some reason. Would it be possible to buid arm64/v8 by using a WM based on Debian 32-bit?