RPi-Distro/pi-gen #646: Unable to Perform Custom Tweaks Within Lite Image

Arszilla, 2022-11-09:
I am attempting to install some custom packages and tweaks for my custom Raspberry Pi OS for my Raspberry Pi Zero and Raspberry Pi 4, while generating a lite image, as I do not need a DE etc. The images are customized using a config like the one below:

```
IMG_NAME="raspbian"
DEPLOY_COMPRESSION="xz"
TARGET_HOSTNAME="raspbian-pi4"
KEYBOARD_KEYMAP="us"
KEYBOARD_LAYOUT="English (US)"
TIMEZONE_DEFAULT="Europe/Tallinn"
FIRST_USER_NAME="arszilla"
DISABLE_FIRST_BOOT_USER_RENAME=1
ENABLE_SSH=1
PUBKEY_SSH_FIRST_USER="ssh-ed25519 REDACTED"
FIRST_USER_PASS="REDACTED"
```

Inside of `stage2`, I have a folder named `04-custom-tweaks. The layout of the `stage2` is as follows:

```
$ tree -a
.
├── 01-sys-tweaks
│   ├── 00-debconf
│   ├── 00-packages
│   ├── 00-packages-nr
│   ├── 00-patches
│   │   ├── 01-useradd.diff
│   │   ├── 02-swap.diff
│   │   ├── 04-inputrc.diff
│   │   ├── 05-path.diff
│   │   ├── 07-resize-init.diff
│   │   └── series
│   ├── 01-run.sh
│   └── files
│       ├── 50raspi
│       ├── 90-qemu.rules
│       ├── console-setup
│       ├── rc.local
│       ├── resize2fs_once
│       └── ttyoutput.conf
├── 02-net-tweaks
│   ├── 00-packages
│   ├── 01-run.sh
│   └── files
│       └── wpa_supplicant.conf
├── 03-accept-mathematica-eula
│   └── 00-debconf
├── 03-set-timezone
│   └── 02-run.sh
├── 04-custom-tweaks
│   ├── 00-packages
│   └── 01-run.sh
├── EXPORT_IMAGE
├── EXPORT_NOOBS
└── prerun.sh
```

As seen above, `04-custom-tweaks` contains `packages` and `01-run.sh`, similar to how the other folders are structured. The custom packages I am installing are as follows:

```
curl
docker.io
git
htop
locate
neofetch
tmux
wget
xdg-user-dirs
zsh
```

Since I am installing `docker.io` and `zsh`, I am attempting to add my `FIRST_USER_NAME` to the `docker` group, and install `oh-my-zsh` to both `FIRST_USER_NAME` and `root`, with the following `01-run.sh` script:

```bash
#!/bin/bash -e

on_chroot << EOF
# If the docker group exists, add the FIRST_USER_NAME to it
if getent group docker >/dev/null; then
    usermod -aG docker ${FIRST_USER_NAME}
fi

# Replace the default shell for all users to zsh
chsh -s /usr/bin/zsh ${FIRST_USER_NAME}
chsh -s /usr/bin/zsh root

# Install oh-my-zsh 
sudo -u ${FIRST_USER_NAME} sh -c "$(curl -fsSL https://raw.githubusercontent.com/ohmyzsh/ohmyzsh/master/tools/install.sh)" "" --unattended
sudo -u ${FIRST_USER_NAME} git clone https://github.com/zsh-users/zsh-autosuggestions /home/${FIRST_USER_NAME}/.oh-my-zsh/plugins/zsh-autosuggestions
sudo -u ${FIRST_USER_NAME} git clone https://github.com/zsh-users/zsh-completions /home/${FIRST_USER_NAME}/.oh-my-zsh/plugins/zsh-completions
sudo -u ${FIRST_USER_NAME} wget -O /home/${FIRST_USER_NAME}/.zshrc https://gitlab.com/Arszilla/dotfiles/-/raw/main/etc/skel/.zshrc

# Install oh-my-zsh for root
sudo -u root sh -c "$(curl -fsSL https://raw.githubusercontent.com/ohmyzsh/ohmyzsh/master/tools/install.sh)" "" --unattended
sudo -u root git clone https://github.com/zsh-users/zsh-autosuggestions /root/.oh-my-zsh/plugins/zsh-autosuggestions
sudo -u root git clone https://github.com/zsh-users/zsh-completions /root/.oh-my-zsh/plugins/zsh-completions
sudo -u root wget -O /root/.zshrc https://gitlab.com/Arszilla/dotfiles/-/raw/main/etc/skel/.zshrc
EOF
```

This is based on the documentation and the trial & errors I have had, but so far, none of the tweaks above (`usermod`, `chsh`, `oh-my-zsh` installation etc.) is happening, and I can't seem to figure out why.

Can I request your assistance on how can I fix this so I can have my images built the way I need them?

Thanks!

XECDesign, 2022-11-10:
I think the issue is your use of `$(...)`. Escape it so that it's `\$(...)`.

Arszilla, 2022-11-10:
That should not be the issue, as seen in [stage1/01-sys-tweaks/00-run.sh](https://github.com/RPi-Distro/pi-gen/blob/c3083ecd503629eac5184ec692f65bbbd28ac317/stage1/01-sys-tweaks/00-run.sh#L8)

XECDesign, 2022-11-10:
Take an extra moment to think about this.

Arszilla, 2022-11-10:
Correct me if I am wrong, but in your statement above, you've said:
> I think the issue is your use of $(...). Escape it so that it's \$(...).

This statement means that my usage of `$(...)` (which doesn't exist in my script, as I used `${...}`) is the cause, and I should use `\` to escape it, which is inapplicable.

I did try using:
- `${FIRST_USER_NAME}`
- `"${FIRST_USER_NAME}"`
- `\${FIRST_USER_NAME}`

All failed with the same error:

```
[15:41:42] Begin /home/arszilla/custom_raspbian/stage2/04-custom-tweaks/01-run.sh
/bin/bash: line 11: fetch: command not found
sudo: unable to resolve host kali-dell: Name or service not known
usage: sudo -h | -K | -k | -V
usage: sudo -v [-AknS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-AknS] [-g group] [-h host] [-p prompt] [-U user] [-u user] [command]
usage: sudo [-AbEHknPS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T
            timeout] [-u user] [VAR=value] [-i|-s] [<command>]
usage: sudo -e [-AknS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T
            timeout] [-u user] file ...
/bin/bash: line 11: -v: command not found
/bin/bash: line 11: -n: command not found
sudo: unable to resolve host kali-dell: Name or service not known
usage: sudo -h | -K | -k | -V
usage: sudo -v [-AknS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-AknS] [-g group] [-h host] [-p prompt] [-U user] [-u user] [command]
usage: sudo [-AbEHknPS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T
            timeout] [-u user] [VAR=value] [-i|-s] [<command>]
usage: sudo -e [-AknS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T
            timeout] [-u user] file ...
sudo: unable to resolve host kali-dell: Name or service not known
usage: sudo -h | -K | -k | -V
usage: sudo -v [-AknS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-AknS] [-g group] [-h host] [-p prompt] [-U user] [-u user] [command]
usage: sudo [-AbEHknPS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T
            timeout] [-u user] [VAR=value] [-i|-s] [<command>]
usage: sudo -e [-AknS] [-r role] [-t type] [-C num] [-D directory] [-g group] [-h host] [-p prompt] [-R directory] [-T
            timeout] [-u user] file ...
/bin/bash: line 11: ...: command not found
/bin/bash: line 11: ...: command not found
/bin/bash: line 11: 033[2m%s033[22m: command not found
/bin/bash: line 11: fg: no job control
sudo: unable to resolve host kali-dell: Name or service not known
/bin/bash: line 242: $'\E[38;2;255;0;0m': command not found
```

XECDesign, 2022-11-10:
You have curly brackets here:  `$(curl -fsSL https://raw.githubusercontent.com/ohmyzsh/ohmyzsh/master/tools/install.sh)`

Arszilla, 2022-11-10:
I am utterly stupid for not realizing the curl command. I sincerely apologize if I came off rude as I totally missed that line. I was too focused on `${FIRST_USER_NAME}` for some reason, thinking the issue was there.

Thanks for the pair of eyes @XECDesign, and apologies once again.

closed 2022-11-10 by Arszilla

XECDesign, 2022-11-10:
No, no problem at all. Hope it all works now.