RPi-Distro/pi-gen #652: Build error in CI `mount: /pi-gen/work/Raspbian/tmpimage/boot: failed to setup loop device`

benfrancis, 2022-11-30:
When running build-docker.sh in CI (inside a [GitHub hosted runner](https://docs.github.com/en/actions/using-github-hosted-runners/about-github-hosted-runners)), I [get the following error](https://github.com/WebThingsIO/gateway/issues/3019):

`mount: /pi-gen/work/Raspbian/tmpimage/boot: failed to setup loop device for /pi-gen/work/Raspbian/export-image/2022-11-30-Raspbian.img.`

This appears to be triggered by this line https://github.com/RPi-Distro/pi-gen/blob/buster/scripts/qcow2_handling#L233

This works when running locally in Ubuntu, but not in GitHub's ubuntu-latest VM.

Do you have any suggestions what might be different about the VM which is causing this to fail? (e.g. a missing kernel module).

More complete output:

```
Image committed.
Rebasing back image-stage1.qcow2 onto image-stage0.qcow2
Rebasing back image-stage2.qcow2 onto image-stage1.qcow2
Rebasing back image-stage3.qcow2 onto image-stage2.qcow2
Mounting image /pi-gen/work/Raspbian/export-image/2022-11-30-Raspbian.qcow2 to rootfs /pi-gen/work/Raspbian/export-image/rootfs
mount: /dev/mapper/nbd0p2 mounted on /pi-gen/work/Raspbian/export-image/rootfs.
mount: /dev/mapper/nbd0p1 mounted on /pi-gen/work/Raspbian/export-image/rootfs/boot.
[17:31:32] Begin /pi-gen/export-image
[17:31:32] Begin /pi-gen/export-image/prerun.sh
[17:31:32] End /pi-gen/export-image/prerun.sh
[17:31:32] Begin /pi-gen/export-image/00-allow-rerun
[17:31:32] Begin /pi-gen/export-image/00-allow-rerun/00-run.sh
[17:31:32] End /pi-gen/export-image/00-allow-rerun/00-run.sh
[17:31:32] End /pi-gen/export-image/00-allow-rerun
[17:31:32] Begin /pi-gen/export-image/01-set-sources
[17:31:32] Begin /pi-gen/export-image/01-set-sources/01-run.sh
Get:1 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:3 http://archive.raspberrypi.org/debian buster/main armhf Packages [392 kB]
Get:4 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:5 http://raspbian.raspberrypi.org/raspbian buster/contrib armhf Packages [58.8 kB]
Get:6 http://raspbian.raspberrypi.org/raspbian buster/non-free armhf Packages [104 kB]
Get:7 http://raspbian.raspberrypi.org/raspbian buster/rpi armhf Packages [1360 B]
Fetched 13.6 MB in 12s (1106 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
[17:31:53] End /pi-gen/export-image/01-set-sources/01-run.sh
[17:31:53] End /pi-gen/export-image/01-set-sources
[17:31:53] Begin /pi-gen/export-image/02-network
[17:31:53] Begin /pi-gen/export-image/02-network/01-run.sh
[17:31:53] End /pi-gen/export-image/02-network/01-run.sh
[17:31:53] End /pi-gen/export-image/02-network
[17:31:53] Begin /pi-gen/export-image/03-set-partuuid
[17:31:53] Begin /pi-gen/export-image/03-set-partuuid/00-run.sh
[17:31:53] End /pi-gen/export-image/03-set-partuuid/00-run.sh
[17:31:53] End /pi-gen/export-image/03-set-partuuid
[17:31:53] Begin /pi-gen/export-image/04-finalise
[17:31:53] Begin /pi-gen/export-image/04-finalise/01-run.sh
Stopping fake hwclock: saving system time.
Mode:     real
Files:    3824
Linked:   791 files
Compared: 0 xattrs
Compared: 1497 files
Saved:    51.03 MiB
Duration: 0.93 seconds
/pi-gen/work/Raspbian/export-image/rootfs: 9.2 GiB (9852047360 bytes) trimmed
/pi-gen/work/Raspbian/export-image/rootfs/sys
/pi-gen/work/Raspbian/export-image/rootfs/proc
/pi-gen/work/Raspbian/export-image/rootfs/dev/pts
/pi-gen/work/Raspbian/export-image/rootfs/dev
/pi-gen/work/Raspbian/export-image/rootfs/boot
/pi-gen/work/Raspbian/export-image/rootfs
/dev/nbd0 disconnected
Connect block device to source qcow2
Resize fs and partition
e2fsck 1.44.5 (15-Dec-2018)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
rootfs: 108757/771552 files (0.1% non-contiguous), 657972/3080192 blocks
resize2fs 1.44.5 (15-Dec-2018)
dumpe2fs 1.44.5 (15-Dec-2018)
resize2fs 1.44.5 (15-Dec-2018)
Resizing the filesystem on /dev/mapper/nbd0p2 to 694080 (4k) blocks.
Begin pass 2 (max = 481780)
Relocating blocks             ----------------------------------------����������������������������������������XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 3 (max = 94)
Scanning inode table          ----------------------------------------����������������������������������������XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Begin pass 4 (max = 16077)
Updating inode references     ----------------------------------------����������������������������������������XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
The filesystem on /dev/mapper/nbd0p2 is now 694080 (4k) blocks long.

Disconnect block device
/dev/nbd0 disconnected
Shrinking qcow2 image
Image resized.
Convert qcow2 to raw image
Get PARTUUIDs from image
Boot: 0164c62c-01
Root: 0164c62c-02
Mount image
mount: /pi-gen/work/Raspbian/tmpimage/boot: failed to setup loop device for /pi-gen/work/Raspbian/export-image/2022-11-30-Raspbian.img.

real	99m17.232s
user	0m0.555s
sys	0m0.460s
Error: Process completed with exit code 1.
```

XECDesign, 2022-12-01:
Does it work if you don't use the qcow2 mechanism? You'd probably need to disable the later stages to build everything within disk space given by github. The qcow2 mechanism is going to be removed at some point.

benfrancis, 2022-12-01:
Weirdly when I re-ran the job the build completed successfully so it must have been something weird with GitHub (I have new errors to fix now). Sorry for the noise.

> The qcow2 mechanism is going to be removed at some point.

That's interesting to know, is there somewhere I can read more about that?

closed 2022-12-01 by benfrancis