RPi-Distro/pi-gen #653: Failure trying to run: chroot "/pi-gen/work/first_test/stage0/rootfs" /bin/true

alexander-sony, 2022-12-02:
I'm trying out pi-gen for the first time, and starting with the simplest config, but get chroot error.

Reproduce with:
```
pi-gen$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.5 LTS
Release:        20.04
Codename:       focal

pi-gen$ git log -1
commit 1d13a292a79f151b06ff84823b9e41781120ce6f...

pi-gen$ cat config 
IMG_NAME=first_test

pi-gen$ ./build-docker.sh
Sending build context to Docker daemon  907.3kB
Step 1/6 : ARG BASE_IMAGE=debian:bullseye
Step 2/6 : FROM ${BASE_IMAGE}
 ---> 2d08b9c48d5b
Step 3/6 : ENV DEBIAN_FRONTEND noninteractive
 ---> Using cache
 ---> d7cef9655bd5
Step 4/6 : RUN apt-get -y update &&     apt-get -y install --no-install-recommends         git vim parted         quilt coreutils qemu-user-static debootstrap zerofree zip dosfstools         libarchive-tools libcap2-bin rsync grep udev xz-utils curl xxd file kmod bc        binfmt-support ca-certificates qemu-utils kpartx fdisk gpg pigz    && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> 4c774d7d4c06
Step 5/6 : COPY . /pi-gen/
 ---> Using cache
 ---> 77ffc38cc47e
Step 6/6 : VOLUME [ "/pi-gen/work", "/pi-gen/deploy"]
 ---> Using cache
 ---> fb053980759d
Successfully built fb053980759d
Successfully tagged pi-gen:latest
WARNING: The requested image's platform (linux/386) does not match the detected host platform (linux/amd64) and no specific platform was requested
[09:41:37] Begin /pi-gen
...
I: Extracting util-linux...
I: Extracting zlib1g...
W: Failure trying to run: chroot "/pi-gen/work/first_test/stage0/rootfs" /bin/true
W: See /pi-gen/work/first_test/stage0/rootfs/debootstrap/debootstrap.log for details
rmdir: failed to remove '/pi-gen/work/first_test/stage0/rootfs/debootstrap': Directory not empty
[09:43:58] bootstrap failed: please check /pi-gen/work/first_test/stage0/debootstrap.log

real    2m22,150s
user    0m0,042s
sys     0m0,068s
```

XECDesign, 2022-12-02:
I don't use docker myself, but I think the issue here is that you need to have qemu-user-static installed on the host system.

alexander-sony, 2022-12-02:
Installing qemu-user-static fixed it.
Thanks!

closed 2022-12-02 by alexander-sony