RPi-Distro/pi-gen #654: E: Unable to fetch some archives ... libpentaho-reporting-flow-engine-java_0.9.4-5.1_all.deb

alexander-sony, 2022-12-02:
I'm trying out pi-gen for the first time, and starting with the simplest config. Get issue with an archive and do not really know how to proceed. Have retried several times, but stop at the same issue every time. Probably this is a problem with the machine at IP: 93.93.128.193. Any a suggestions?

apt-get update --fix-missing does not help.

```
pi-gen$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.5 LTS
Release:        20.04
Codename:       focal

pi-gen$ git log -1
commit 1d13a292a79f151b06ff84823b9e41781120ce6f...

pi-gen$ cat config 
IMG_NAME=first_test

pi-gen$ ./build-docker.sh
...
aspell-autobuildhash: processing: en [en_US-wo_accents-only].
[11:49:52] End /pi-gen/stage5/00-install-extras/00-packages
[11:49:52] End /pi-gen/stage5/00-install-extras
[11:49:52] Begin /pi-gen/stage5/00-install-libreoffice
[11:49:52] Begin /pi-gen/stage5/00-install-libreoffice/00-packages
Reading package lists...
...
Get:232 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libreoffice-wiki-publisher all 1.2.0+LibO7.0.4-4+rpi1+deb11u3 [1579 kB]
Ign:196 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libpentaho-reporting-flow-engine-java all 0.9.4-5.1
Ign:196 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libpentaho-reporting-flow-engine-java all 0.9.4-5.1
Err:196 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf libpentaho-reporting-flow-engine-java all 0.9.4-5.1
  Connection failed [IP: 93.93.128.193 80]
E: Fetched 568 MB in 43s (13.1 MB/s)
Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/pentaho-reporting-flow-engine/libpentaho-reporting-flow-engine-java_0.9.4-5.1_all.deb  Connection failed [IP: 93.93.128.193 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

real    34m5,074s
user    0m0,620s
sys     0m0,464s
```

XECDesign, 2022-12-02:
We use apt-cacher-ng and configure `APT_PROXY` accordingly. Aside from improving reliability, you could also configure it to redirect to a local or more reliable mirror.

alexander-sony, 2022-12-05:
> We use apt-cacher-ng and configure `APT_PROXY` accordingly. Aside from improving reliability, you could also configure it to redirect to a local or more reliable mirror.

Got it.

closed 2022-12-05 by alexander-sony