RPi-Distro/pi-gen #657: Running in chroot, ignoring command 'start' during stage2/02-net-tweaks

Mawiguk0, 2022-12-09:
Log Output:
```log
[13:01:27] End /root/mypi/stage2/02-net-tweaks/00-packages
[13:01:27] Begin /root/mypi/stage2/02-net-tweaks/01-run.sh
'files/wpa_supplicant.conf' -> '/root/my-pi/work/my-pi/stage2/rootfs/etc/wpa_supplicant/wpa_supplicant.conf'
Running in chroot, ignoring command 'start'

```
The config file is pretty plain

```config
# Config used to build my-pi images for RPi3 and RPi4

IMG_NAME="my-pi"
RELEASE="bullseye"
LOCALE_DEFAULT="de_DE.UTF-8"
TIMEZONE_DEFAULT="Europe/London"
FIRST_USER_NAME="redacted"
#FIRST_USER_PASS="redacted"
FIRST_USER_PASS="redacted"
STAGE_LIST="stage0 stage1 stage2"
DEPLOY_COMPRESSION=none

TARGET_HOSTNAME=my-pi
KEYBOARD_KEYMAP="de"
DISABLE_FIRST_BOOT_USER_RENAME=1

WPA_ESSID="redacted"
WPA_PASSWORD="redacted"
WPA_COUNTRY="de"
```

It fails during this execution in
stage2/02-net-tweaks/01-run.sh line: 11-15
```bash
if [ -v WPA_COUNTRY ]; then
        on_chroot <<- EOF
                SUDO_USER="${FIRST_USER_NAME}" raspi-config nonint do_wifi_country "${WPA_COUNTRY}"
        EOF
fi
```

### Reproduce:
clone and build pi-gen

Mawiguk0, 2022-12-13:
@XECDesign just pinging you up because of your activity in this repo. Do you know who to ask or ping for this issue?

XECDesign, 2022-12-14:
Hey, no it's alright you've pinged the right person. It's just that I haven't gotten around to taking a proper look at this yet. It sounds like raspi-config returns false when it shouldn't.

XECDesign, 2022-12-21:
Does it work now if you change `WPA_COUNTRY="de"` to `WPA_COUNTRY="DE"`?

Mawiguk0, 2022-12-23:
Same behavior with WPA_COUNTRY="DE" and WPA_COUNTRY="EN" or WPA_COUNTRY="en"

I just ran the build process with WPA_COUNTRY not set to anything by commenting it. The build is running just fine without that.

Mawiguk0, 2022-12-23:
I guess this issue from raspi-config is relating to this behavior:
[raspi-config issue 160](https://github.com/RPi-Distro/raspi-config/issues/160)

XECDesign, 2022-12-23:
Nope that issue is unrelated. 

Is this a clean build or are you continuing a previous one?

I recently uploaded a new raspi-config which fixed the issue for me.

I’ll double check when I get a chance.

Mawiguk0, 2023-01-26:
We can close this issue i guess^^

I just Raspberry Pi OS without any issues on a new build machine...

closed 2023-01-26 by Mawiguk0

Morrigan-Ship, 2023-09-08:
same problem systemctl not work chroot

XECDesign, 2023-09-08:
> same problem systemctl not work chroot

It's not meant to.