RPi-Distro/pi-gen #661: Error while trying to build pi-gen on Ubuntu 22.10

PRochefort, 2023-01-13:
Hi,

I used to be able to build this without much trouble a while back but recently, I have been trying to build this on an Ubuntu 22.10 but it always fail. The error I get is this:

W: Failure trying to run: chroot "/home/pierre/pi-gen/work/Raspbian/stage0/rootfs" /bin/true
W: See /home/pierre/pi-gen/work/Raspbian/stage0/rootfs/debootstrap/debootstrap.log for details
rmdir: failed to remove '/home/pierre/pi-gen/work/Raspbian/stage0/rootfs/debootstrap': Directory not empty

Can someone point me in the right direction to get this fixed? I know I probably don't need to build this myself but I wanted to try it. It used to work flawlessly for the longest time but now, I can't figure this out.

I'm running the ./build.sh with a sudo command if that can help.

Thank you 
Pierre

XECDesign, 2023-01-17:
Is there more to the error message? What's in `debootstrap.log`? Have there been changes to your build environment between the time it was working and now?

PRochefort, 2023-01-18:
Hi,

Thank you for taking the time to post.

Sorry about the late response. As far as I know that's the only error message but let me look for the log file you are referring to and I will attach it here. It might take a bit of time but I will get it eventually.

As for the build environment, it's a brand new one. I had built this in a VM before using an Ubuntu 20.04 or something similar but I didn't keep the VM - a dumb mistake on my part now that I look back on it. I started from scratch with a stock 22.10. When I saw that it wasn't working, I remembered that this was tested on a 16.04 so I downloaded that and built a VM with it but it didn't seem to really help.

As soon as I have more, I will update this.
Thank you.

XECDesign, 2023-01-19:
From the information so far, it seems like a problem with the version or setup of qemu in 22.10. The only distro I can currently vouch for as suitable for pi-gen cross-builds would be Debian Bullseye.

I'm currently on 22.04 myself, and haven't noticed any issues with it.

PRochefort, 2023-01-23:
I will be closing this. As mentioned, I did the same build on a 22.04 instead and it worked perfectly. 
The error is likely mine to start with. Thanks for those that answered the post.
Have a great day.

closed 2023-01-23 by PRochefort