RPi-Distro/pi-gen #662: linux-perf package is 5.10 for Debian, but Raspberry Pi kernel is 5.15.

JinShil, 2023-01-16:
Forgive me if this is not the right place to report this, but there is a version mismatch that prevents installing `linux-perf` for the current release of the Raspberry Pi OS.  The Raspberry Pi OS is on kernel version 5.15, but Debian is on kernel version 5.10. Therefore, when one tries to run `sudo apt install linux-perf` it installs `linux-perf-5.10`.  Then when running `perf` it fails with `/usr/bin/perf: line 13: exec: perf_5.15: not found`.

Please consider adding a `linux-perf` package that matches the Raspberry Pi kernel version to the Raspberry Pi repository.

XECDesign, 2023-01-17:
This will be part of the work of migrating away from the raspberrypi-firmware package to upstream debian ones where possible. Aiming to have that done for the Bookworm release.

JinShil, 2024-06-18:
This appears to have been resolved.

closed 2024-06-18 by JinShil