RPi-Distro/pi-gen #665: How to lock package versions.

murrellr, 2023-01-19:
I am using pi-gen to build a custom image for a PI Compute Module 4 to be used in an industrial product.  As the build is running, I see many packages being loaded by Apt from remote locations.  For product stability and tracking, once we release the product, we want to lock the versions of all the packages.  If I rebuild the image at a later date, I want to use the package versions from the original release of our software, not the latest.  I would only want to use the latest package versions when we do a major release, not a minor one.

Is it possible to do this?

murrellr, 2023-01-19:
I know of a package called reprepro that creates a local apt repositories from remote ones to speed up access.  Perhaps between the two I can achieve this.  Thanks for the information.

svet-b, 2023-04-21:
Wanted to piggy-back on this, as I have a similar question. Though in my case I would like to lock the versions of just a couple of packages, rather than all. Specifically, the kernel and firmware packages. [The context is that we have an upgrade mechanism where the root partition is upgraded but boot is usually left intact. So I'd like to be able to generate new root partitions that match the kernel/firmware version in previously created boot partitions]

A couple of approaches I'm considering:
1. Pinning the package versions via a config in `/etc/apt/preferences` (as suggested in https://serverfault.com/a/435416)
2. "Vendoring" the relevant .debs (at the desired version) within my repo and installing them directly with `dpkg` rather than `apt`

The first approach seems more elegant - but I guess prone to failure in the longer run, if/when the relevant package versions are no longer available from the respective APT repos. I suppose I could work around that by using a snapshot repo (or mirroring locally) as @XECDesign suggested, but that's probably not a justified level of complexity if I'm only doing it for the sake of a couple of packages.

Would approach #2 then make more sense? Or am I likely to end up in some kind of dependency hell? (in my experience most packages don't depend strongly on kernel version but who knows)

Also open to other suggestions of course!

nullromo, 2025-01-15:
I have the same problem, and the way I wanted to solve it was by just creating an offline apt repo.

<details>

<summary>This is what I did first to generate the offline repo</summary>

1. I figured out all the packages that pi-gen was installing and downloaded them all using `apt-get download` and/or `apt-get install --reinstall --download-only`.
2. I used `dpkg-scanpackages` to create a `Packages` file.
3. In `pi-gen/stage0/00-configure-apt/00-run.sh`, I copied the repo to `/usr/local/local-apt-repo`.
4. I pointed `pi-gen/stage0/00-configure-apt/files/sources.list` to `deb [trusted=yes] file:/usr/local/local-apt-repo ./`
5. When running the pi-gen build, it began pulling packages from the local repo instead of from `raspbian.raspberrypi.com/raspbian`
6. I believe this apt repo structure is called the "[trivial style](https://wiki.debian.org/DebianRepository/Setup)"

</details>

After that, things were working okay, but I realized the debootstrap step was still pulling packages from the internet during the `prerun.sh` script of stage 0.

So, I tried to have debootstrap use the local repo, but that's where I ran into some problems. I restructured the structure of my local repo according to [this guide](https://earthly.dev/blog/creating-and-hosting-your-own-deb-packages-and-apt-repo/). Now when I run the build, I get the following in the debootstrap log:

```
armhf: ok
mknod: /pi-gen/work/Dicon-Raspbian/stage0/rootfs/dev/null: No such file or directory
```

So it appears debootstrap is failing to create `/dev/null`, and if I manually create `/dev/null` then it fails to create `/usr` and `/sys`, etc.

So is there something that needs to be in the local repo that tells debootstrap to create these directories?

nullromo, 2025-01-17:
Actually I was able to use Aptly to build the repo correctly.

Here are the steps I used:

0. [Install aptly](https://www.aptly.info/download/).
1. Get keys
    1. `wget -O - http://raspbian.raspberrypi.org/raspbian.public.key | gpg --no-default-keyring --keyring trustedkeys.gpg --import`
    1. `wget -O - https://archive.raspberrypi.com/debian/raspberrypi.gpg.key | gpg --no-default-keyring --keyring trustedkeys.gpg --import`
1. Create mirrors
    1. `aptly mirror create raspbian-mirror http://raspbian.raspberrypi.org/raspbian/ bookworm`
    1. `aptly mirror create debian-mirror https://archive.raspberrypi.com/debian/ bookworm`
1. Update mirrors
    1. `aptly mirror update raspbian-mirror` (NOTE: this will take a long time, maybe 6-8 hours)
    1. `aptly mirror update debian-mirror`
1. Create repo `aptly repo create ossgp-repo`
1. Import packages from the mirrors into the repo, something like this, but adjusting as necessary. There are a ton of packages that need to be added, but I didn't want to paste them all in here.
```
#!/bin/bash

set -e

RASPBIAN_PACKAGES="locales initramfs-tools adduser apt apt-utils base-files" # ..............
DEBIAN_PAKAGES="raspberrypi-archive-keyring linux-image-rpi-v6 linux-image-rpi-v7" # ..............

for ITEM in $RASPBIAN_PACKAGES; do
# for ITEM in $DEBIAN_PACKAGES; do
    echo "Looking for $ITEM"
    aptly repo import -with-deps raspbian-mirror ossgp-repo "Name (% ${ITEM})"
    # aptly repo import -with-deps debian-mirror ossgp-repo "Name (% ${ITEM})"
done
```

6. Publish the repo `aptly publish repo -distribution="bookworm" -skip-signing ossgp-repo`
    1. In my case I figured it was ok to use `-skip-signing` since I'm just using this locally for pi-gen and never using it after the image is built, so there is no network involved at all.
7. Results are in `~/.aptly/public`

Hopefully that helps anyone else who comes across this.