RPi-Distro/pi-gen #668: NO_PUBKEY 82B129927FA3303E

brianmay, 2023-02-05:
```
[20:17:32] Begin /home/brian/tree/3rdparty/pi-gen
[20:17:32] Begin /home/brian/tree/3rdparty/pi-gen/stage0
[20:17:32] Begin /home/brian/tree/3rdparty/pi-gen/stage0/prerun.sh
[20:17:32] End /home/brian/tree/3rdparty/pi-gen/stage0/prerun.sh
[20:17:32] Begin /home/brian/tree/3rdparty/pi-gen/stage0/00-configure-apt
[20:17:32] Begin /home/brian/tree/3rdparty/pi-gen/stage0/00-configure-apt/00-run.sh
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Get:2 http://archive.raspberrypi.org/debian bullseye InRelease [23.6 kB]
Err:2 http://archive.raspberrypi.org/debian bullseye InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 82B129927FA3303E
Reading package lists... Done
W: http://raspbian.raspberrypi.org/raspbian/dists/bullseye/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/raspberrypi-archive-stable.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: http://archive.raspberrypi.org/debian/dists/bullseye/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/raspberrypi-archive-stable.gpg are ignored as the file is not readable by user '_apt' executing apt-key.
W: GPG error: http://archive.raspberrypi.org/debian bullseye InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 82B129927FA3303E
E: The repository 'http://archive.raspberrypi.org/debian bullseye InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

I feel like I am doing something stupid here, but I can't see anything wrong.

raspbian-archive-keyring was installed.

XECDesign, 2023-02-05:
What's the build environment and filesystem?

> The key(s) in the keyring /etc/apt/trusted.gpg.d/raspberrypi-archive-stable.gpg are ignored as the file is not readable by user '_apt' executing apt-key.

I'd start by looking at that file (in the work directory), not the host system.  What's the content and who is it owned by?

brianmay, 2023-02-05:
Oh, crap. I diagnosed this before, but then totally stuffed up trying to reset the state. I hadn't noticed that directories like `/dev` were still mounted in the work directory, so when I `rm -rf work` it obviously deleted all my `/dev` files :-(, and I had to reboot, and then I completely forgot. Anyway:

```
canidae# ls -l ./work/brian-slint/stage0/rootfs/etc/apt/trusted.gpg.d/raspberrypi-archive-stable.gpg
-rw------- 1 root root 1183 Feb  6 08:53 ./work/brian-slint/stage0/rootfs/etc/apt/trusted.gpg.d/raspberrypi-archive-stable.gpg

canidae# gpg < ./work/brian-slint/stage0/rootfs/etc/apt/trusted.gpg.d/raspberrypi-archive-stable.gpg
gpg: WARNING: no command supplied.  Trying to guess what you mean ...
pub   rsa2048 2012-06-17 [SC]
      CF8A1AF502A2AA2D763BAE7E82B129927FA3303E
uid           Raspberry Pi Archive Signing Key
sub   rsa2048 2012-06-17 [E]
```

My strong suspicion is that the permissions are wrong, because I use `umask 077`. Which probably means that there should be a `umask 000` somewhere in one of the scripts.

brianmay, 2023-02-05:
Setting umask to 000 before running the script does appear to have helped.

XECDesign, 2023-02-06:
What if you change [this](https://github.com/RPi-Distro/pi-gen/blob/master/stage0/00-configure-apt/00-run.sh#L15) to this:

```
cat files/raspberrypi.gpg.key | gpg --dearmor > "${STAGE_WORK_DIR}/raspberrypi-archive-stable.gpg"
install -m 644 "${STAGE_WORK_DIR}/raspberrypi-archive-stable.gpg" "${ROOTFS_DIR}/etc/apt/trusted.gpg.d/"
```

brianmay, 2023-02-07:
Yes, that seems to work also.

XECDesign, 2023-02-09:
Should be fixed now

closed 2025-04-07 by XECDesign