RPi-Distro/pi-gen #669: FAT partition label: don't name it "boot"

macmpi, 2023-02-13:
There is a quite maddening side effect of [default `boot` partition label](https://github.com/RPi-Distro/pi-gen/blob/master/export-noobs/00-release/files/partitions.json#L5) together with a longstanding firmware bug uncovered by @pelwell :
https://github.com/raspberrypi/firmware/issues/1529#issuecomment-774102402

That partition naming prevents use of `boot` directory name inside it, for anything essential to boot the system from FAT partition... Not a major problem for PiOS but has adverse consequence for other distribution that may be installed after.

Agreed, root of problem is in firmware side, but as this does not seem to get resolved, here is a typical consequential use-case if default Pi OS image use that `boot` label.
- User makes an initial Pi OS image media (with `rpi-imager` or manually), and runs fine on his device: all good.
- Then user may want to try another distribution for Pi device (Alpine to name one), which has a `boot` directory inside FAT partition.
- User replaces FAT partition content with new distribution's files
- Device won't boot.
- Try default reformat, re-install...won't boot either

All this because first-off media generation had that colliding `boot` label... definitely not an obvious guess!

Would it be possible to consider another non-colliding label name for that default FAT partition (`FATroot` or something), until that firmware bug is eventually fixed?

XECDesign, 2023-02-13:
@pelwell, @timg236, is this a bootrom issue on older pis, or can it be fixed in bootcode.bin or start.elf files?

pelwell, 2023-02-13:
It's somewhere deep in the filesystem abstractions of start*.elf.

closed 2023-02-14 by XECDesign

macmpi, 2023-02-14:
thanks :+1:

XECDesign, 2023-02-14:
Thanks for reporting the issue

macmpi, 2023-02-14:
Hopefully firmware will get a permanent fix at some point, but at least one main source of catch-22 should be gone with next release. :wink:

commit 5d2c6f31 (2023-02-14): export-image: change boot partition lable to 'bootfs'

Fixes #669