RPi-Distro/pi-gen #671: Is there a reason why the option is not given when using umount? (Target is busy)

bdh1993, 2023-03-02:
If you look at the unmount() function in the "script > common" path, no option is applied.
Therefore, a lot of target is busy situations occur during the build.

You can solve this problem by attaching the "-lf" option when you do umount.

It worked in the local build. (I don't know if it's build from Docker.)

There was no problem when using the generated image.
Is there any other problem I don't know?

XECDesign, 2023-03-02:
Because using `-l` doesn't actually unmount it. It removes the mount entry, but as long as the files remain open, it's still busy and the same as being mounted in the background. It's only when the last reference to all open files is dropped that the umount actually happens. I may be wrong, but that's my understanding, and without having evidence that it's 100% safe, I don't want to risk it.

It would be better to understand why `umount` doesn't work int he first place. I haven't seen that problem in a while, so I assume it was an issue that has been resolved in an upstream package somewhere.

bdh1993, 2023-03-03:
Thank you.
I build it on my computer.
When building, I opened another window or searched the website.
In the process, you have to umount, but I think it's because there's a row hanging in the other process.
Would this issue be more free if build from Docker?

XECDesign, 2023-03-03:
> When building, I opened another window or searched the website.
> In the process, you have to umount, but I think it's because there's a row hanging in the other process.

Sorry, I don't quite get what you mean there.

> I build it on my computer.

Out of curiosity, what distro are you running?

> Would this issue be more free if build from Docker?

I don't know, I don't use docker for my own builds.