RPi-Distro/pi-gen #672: Failure trying to run: chroot "/home/warfull/pi-gen/work/PILITE/stage0/rootfs" /bin/true

Warfull, 2023-03-04:
Last two lines in debootsrap.log  :
2023-03-04 14:02:44 URL:http://mirrors.ircam.fr/pub/raspbian/raspbian/pool/main/z/zlib/zlib1g_1.2.11.dfsg-2+deb11u2_armhf.deb [84816/84816] -> "/home/warfull/pi-gen/work/PILITE/stage0/rootfs//var/cache/apt/archives/partial/zlib1g_1%3a1.2.11.dfsg-2+deb11u2_armhf.deb" [1]
Allocating guest commpage: Cannot allocate memory

Computer : Dell PRECISION M4800
CPU : I7
RAM 16G

Fresh Debian 11 install
qemu-user-static installed...

thirstyice, 2023-06-19:
Maybe related to #271?

wandering-andy, 2023-10-29:
Is `binfmt-support` installed on your machine? I was getting a very similar error message and installing it seems to have resolved the issue. You might also check out [this](https://github.com/RPi-Distro/pi-gen#binfmt_misc) section of the readme.