RPi-Distro/pi-gen #674: Is adding random environment variables to config supported?

maxpavlov, 2023-03-08:
Can I add my own environment variable to `config` and expect it to be available for all the .sh scripts that execute during a build? 
Something like
```
IMG_NAME="Essential-zero" 
RELEASE="bullseye" 
MY_CUSTOM_VARIABLE=1
```

Thanks.

bdh1993, 2023-03-09:
As far as I know, it doesn't matter if you put any environment variables in the config file.

For example, you can view the FIRST_USER_NAME variable in "build.sh".

This variable is specified only for build.sh, but is used in the file "stage2/01sys-tweaks/01-run.sh".

Variables must be "export" from "build.sh."

[image]

maxpavlov, 2023-03-12:
Yes, export works. 
So to get it to work you add a variable to config and 'export' it from the build.sh. Minimum of two actions to make a custom env var available to stage scripts. 

It works well, I'll close this one. Thanks.

closed 2023-03-12 by maxpavlov

MexHigh, 2024-03-05:
How would you need to modify the `build-docker.sh` in order to accomplish this? I want to access a custom variable in a `XX-run-chroot.sh` script