RPi-Distro/pi-gen #678: README: update `binfmt_misc` troubleshooting section for `aarch64`

stefanb2, 2023-03-22:
When running a build from the `arm64` branch, i.e. for the `aarch64` target, then `/usr/bin/qemu-aarch64-static` is required to execute the target binaries on the host machine:

~~~log
# file /pi-gen/work/<IMG_NAME>/stage0/rootfs/bin/echo
/pi-gen/work/<IMG_NAME>/stage0/rootfs/bin/echo: ELF 64-bit LSB pie executable, ARM aarch64, version 1 (SYSV), ...
# ls /proc/sys/fs/binfmt_misc/
qemu-aarch64  qemu-aarch64_be  qemu-arm  qemu-armeb  register  status
# chroot "/pi-gen/work/<IMG_NAME>/stage0/rootfs" /bin/echo it works!
it works!
~~~

Depending on the host system the user may have to install a different package to get QEMU aarch64 support, f.ex. on Fedora it would be `qemu-user-static-aarch64`.

Adding this to the troubleshooting section will hopefully save hours of head-scratching for other users...

XECDesign, 2023-03-27:
In general we only support Debian and some versions of Ubuntu, but I'd accept a PR if you think it'll help others.

stefanb2, 2023-03-27:
Sure, whenever GitHub recovers from the current meltdown...

commit 3e8046da (2023-03-27): Update qemu ARM binary name in README.md (#684)

When running a 64 bit build you need the QEMU aarch64 binary installed
on the host system.

Fixes #678
closed 2023-03-28 by stefanb2