RPi-Distro/pi-gen #691: Can't add new partitions

H4kim, 2023-04-09:
I want to add new partitions instead of resizing the root partition to take all the available space on the disk, I tried to play with these two files: [file1](https://github.com/RPi-Distro/pi-gen/blob/master/export-image/prerun.sh), [file2](https://github.com/RPi-Distro/raspi-config/blob/master/usr/lib/raspi-config/init_resize.sh#L71), but i got no success, i know this is just me not knowing how to do it, and that's why i'm asking for help after 1 week of work.

Thanks a lot for helping me find where do i can make changes to add partitions.

XECDesign, 2023-04-11:
Instead of the linked `file2`, try [this one](https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/master/usr/lib/raspberrypi-sys-mods/firstboot).

o-alquimista, 2024-01-19:
> Instead of the linked `file2`, try [this one](https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/master/usr/lib/raspberrypi-sys-mods/firstboot).

I'm trying to figure out how to do this, and my guess is that I need to somehow patch the "firstboot" script - more specifically the `do_resize()` function, and perform the partitioning there. But how do you patch something that comes from a deb package? I've never done that before.

My first idea was to create another shell script, just like "firstboot", and add `init=/path/to/script` to `cmdline.txt`. The problem is that would create a conflict - two scripts wanting the "first boot".

Any ideas? My goal is to separate `/var` and `/home`, and encrypt them with `cryptsetup`. [I already do this manually](https://gitlab.com/-/snippets/2594880), but it's a lot of work.

XECDesign, 2024-01-29:
You can patch the file like you would any other. Although apt will overwrite it when the package is updated, that won't happen before first boot.

Or if you add your own and make it the init script instead, as long as it does the job of the original init script as well, it should be fine.

closed 2024-01-29 by XECDesign