RPi-Distro/pi-gen #698: How to build image from Tag 2023-05-03-raspios-bullseye?

Psychesnet, 2023-07-19:
Hi pi-gen,

I try to clone repo and try to build my own image, It always fails to strange part.
I just clone repo and run build-docker.sh and the error log are

```
W: Failure trying to run: chroot "/pi-gen/work/KioskPi/stage0/rootfs" /bin/true
W: See /pi-gen/work/KioskPi/stage0/rootfs/debootstrap/debootstrap.log for details
rmdir: failed to remove '/pi-gen/work/KioskPi/stage0/rootfs/debootstrap': Directory not empty
[06:30:18] bootstrap failed: please check /pi-gen/work/KioskPi/stage0/debootstrap.log

real    3m24.033s
user    0m0.070s
sys     0m0.035s
erichsieh@j01:~/workdir/pi-gen$ PRESERVE_CONTAINER=1 CONTINUE=1 ./build-docker.sh
Sending build context to Docker daemon  913.9kB
Step 1/4 : ARG BASE_IMAGE=debian:bullseye
Step 2/4 : FROM ${BASE_IMAGE}
 ---> 985169c5e2de
Step 3/4 : ENV DEBIAN_FRONTEND noninteractive
 ---> Using cache
 ---> 53bbc6222db1
Step 4/4 : RUN apt-get -y update &&     apt-get -y install --no-install-recommends         git vim parted         quilt coreutils qemu-user-static debootstrap zerofree zip dosfstools         libarchive-tools libcap2-bin rsync grep udev xz-utils curl xxd file kmod bc        binfmt-support ca-certificates qemu-utils kpartx fdisk gpg pigz    && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> ee40d3892e09
Successfully built ee40d3892e09
Successfully tagged pi-gen:latest
WARNING: The requested image's platform (linux/386) does not match the detected host platform (linux/amd64) and no specific platform was requested
[06:31:19] Begin /pi-gen
[06:31:19] Begin /pi-gen/stage0
[06:31:19] Begin /pi-gen/stage0/prerun.sh
[06:31:19] End /pi-gen/stage0/prerun.sh
[06:31:19] Begin /pi-gen/stage0/00-configure-apt
[06:31:19] Begin /pi-gen/stage0/00-configure-apt/00-run.sh
install: target '/pi-gen/work/KioskPi/stage0/rootfs/etc/apt/sources.list.d/' is not a directory: No such file or directory

real    0m0.524s
user    0m0.006s
sys     0m0.006s
erichsieh@j01:~/workdir/pi-gen$ PRESERVE_CONTAINER=1 CONTINUE=1 ./build-docker.sh
Sending build context to Docker daemon  913.9kB
Step 1/4 : ARG BASE_IMAGE=debian:bullseye
Step 2/4 : FROM ${BASE_IMAGE}
 ---> 985169c5e2de
Step 3/4 : ENV DEBIAN_FRONTEND noninteractive
 ---> Using cache
 ---> 53bbc6222db1
Step 4/4 : RUN apt-get -y update &&     apt-get -y install --no-install-recommends         git vim parted         quilt coreutils qemu-user-static debootstrap zerofree zip dosfstools         libarchive-tools libcap2-bin rsync grep udev xz-utils curl xxd file kmod bc        binfmt-support ca-certificates qemu-utils kpartx fdisk gpg pigz    && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> ee40d3892e09
Successfully built ee40d3892e09
Successfully tagged pi-gen:latest
WARNING: The requested image's platform (linux/386) does not match the detected host platform (linux/amd64) and no specific platform was requested
[06:31:22] Begin /pi-gen
[06:31:22] Begin /pi-gen/stage0
[06:31:22] Begin /pi-gen/stage0/prerun.sh
[06:31:22] End /pi-gen/stage0/prerun.sh
[06:31:22] Begin /pi-gen/stage0/00-configure-apt
[06:31:22] Begin /pi-gen/stage0/00-configure-apt/00-run.sh
install: target '/pi-gen/work/KioskPi/stage0/rootfs/etc/apt/sources.list.d/' is not a directory: No such file or directory

real    0m0.541s
user    0m0.013s
sys     0m0.000s
erichsieh@j01:~/workdir/pi-gen$ PRESERVE_CONTAINER=1 CONTINUE=1 ./build-docker.sh
Sending build context to Docker daemon  913.9kB
Step 1/4 : ARG BASE_IMAGE=debian:bullseye
Step 2/4 : FROM ${BASE_IMAGE}
 ---> 985169c5e2de
Step 3/4 : ENV DEBIAN_FRONTEND noninteractive
 ---> Using cache
 ---> 53bbc6222db1
Step 4/4 : RUN apt-get -y update &&     apt-get -y install --no-install-recommends         git vim parted         quilt coreutils qemu-user-static debootstrap zerofree zip dosfstools         libarchive-tools libcap2-bin rsync grep udev xz-utils curl xxd file kmod bc        binfmt-support ca-certificates qemu-utils kpartx fdisk gpg pigz    && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> ee40d3892e09
Successfully built ee40d3892e09
Successfully tagged pi-gen:latest
WARNING: The requested image's platform (linux/386) does not match the detected host platform (linux/amd64) and no specific platform was requested
[06:31:46] Begin /pi-gen
[06:31:46] Begin /pi-gen/stage0
[06:31:46] Begin /pi-gen/stage0/prerun.sh
[06:31:46] End /pi-gen/stage0/prerun.sh
[06:31:46] Begin /pi-gen/stage0/00-configure-apt
[06:31:46] Begin /pi-gen/stage0/00-configure-apt/00-run.sh
install: target '/pi-gen/work/KioskPi/stage0/rootfs/etc/apt/trusted.gpg.d/' is not a directory: No such file or directory

real    0m0.539s
user    0m0.005s
sys     0m0.010s
erichsieh@j01:~/workdir/pi-gen$ PRESERVE_CONTAINER=1 CONTINUE=1 ./build-docker.sh
Sending build context to Docker daemon  913.9kB
Step 1/4 : ARG BASE_IMAGE=debian:bullseye
Step 2/4 : FROM ${BASE_IMAGE}
 ---> 985169c5e2de
Step 3/4 : ENV DEBIAN_FRONTEND noninteractive
 ---> Using cache
 ---> 53bbc6222db1
Step 4/4 : RUN apt-get -y update &&     apt-get -y install --no-install-recommends         git vim parted         quilt coreutils qemu-user-static debootstrap zerofree zip dosfstools         libarchive-tools libcap2-bin rsync grep udev xz-utils curl xxd file kmod bc        binfmt-support ca-certificates qemu-utils kpartx fdisk gpg pigz    && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> ee40d3892e09
Successfully built ee40d3892e09
Successfully tagged pi-gen:latest
WARNING: The requested image's platform (linux/386) does not match the detected host platform (linux/amd64) and no specific platform was requested
[06:31:58] Begin /pi-gen
[06:31:58] Begin /pi-gen/stage0
[06:31:58] Begin /pi-gen/stage0/prerun.sh
[06:31:58] End /pi-gen/stage0/prerun.sh
[06:31:58] Begin /pi-gen/stage0/00-configure-apt
[06:31:58] Begin /pi-gen/stage0/00-configure-apt/00-run.sh
execve '/bin/bash' failed!

real    0m0.555s
user    0m0.006s
sys     0m0.006s

```

Anyone can help? Thanks.

XECDesign, 2023-07-19:
At a glance, it looks like a binfmt/qemu issue - the docket container isn't able to run arm binaries. Can you post that debootstrap.log file?

It looks like that's where it initially failed and because you're not doing a clean build, it sees that there are files there already and doesn't try to run debootstrap again. Since the rest of the build doesn't see the files that should be there, it fails on future runs.

Psychesnet, 2023-07-20:
Hi @XECDesign 

Thanks for replying so quickly.

I remove whole folder and git clone again, then ./build-docker.sh to try, I don't see debootstrap.log
What should I do to have the log file? Thanks.
```
root@36869b33627e:/pi-gen# find . -name *.log
./work/KioskPi/build.log
./work/KioskPi/stage2/rootfs/var/log/bootstrap.log
./work/KioskPi/stage2/rootfs/var/log/apt/term.log
./work/KioskPi/stage2/rootfs/var/log/apt/history.log
./work/KioskPi/stage2/rootfs/var/log/alternatives.log
./work/KioskPi/stage2/rootfs/var/log/dpkg.log
./work/KioskPi/stage1/rootfs/var/log/bootstrap.log
./work/KioskPi/stage1/rootfs/var/log/apt/term.log
./work/KioskPi/stage1/rootfs/var/log/apt/history.log
./work/KioskPi/stage1/rootfs/var/log/alternatives.log
./work/KioskPi/stage1/rootfs/var/log/dpkg.log
./work/KioskPi/stage0/rootfs/var/log/bootstrap.log
./work/KioskPi/stage0/rootfs/var/log/apt/term.log
./work/KioskPi/stage0/rootfs/var/log/apt/history.log
./work/KioskPi/stage0/rootfs/var/log/alternatives.log
./work/KioskPi/stage0/rootfs/var/log/dpkg.log
./work/KioskPi/stage5/rootfs/usr/share/doc/python3.9/pybench.log
./work/KioskPi/stage5/rootfs/usr/share/doc/openjdk-11-jre-headless/test-armhf/jtreg_output-hotspot.log
./work/KioskPi/stage5/rootfs/var/log/bootstrap.log
./work/KioskPi/stage5/rootfs/var/log/apt/term.log
./work/KioskPi/stage5/rootfs/var/log/apt/history.log
./work/KioskPi/stage5/rootfs/var/log/alternatives.log
./work/KioskPi/stage5/rootfs/var/log/dpkg.log
./work/KioskPi/stage5/rootfs/var/log/fontconfig.log
./work/KioskPi/stage4/rootfs/usr/share/doc/python3.9/pybench.log
./work/KioskPi/stage4/rootfs/var/log/bootstrap.log
./work/KioskPi/stage4/rootfs/var/log/apt/term.log
./work/KioskPi/stage4/rootfs/var/log/apt/history.log
./work/KioskPi/stage4/rootfs/var/log/alternatives.log
./work/KioskPi/stage4/rootfs/var/log/dpkg.log
./work/KioskPi/stage4/rootfs/var/log/fontconfig.log
./work/KioskPi/stage3/rootfs/var/log/bootstrap.log
./work/KioskPi/stage3/rootfs/var/log/apt/term.log
./work/KioskPi/stage3/rootfs/var/log/apt/history.log
./work/KioskPi/stage3/rootfs/var/log/alternatives.log
./work/KioskPi/stage3/rootfs/var/log/dpkg.log
./work/KioskPi/stage3/rootfs/var/log/fontconfig.log
```

XECDesign, 2023-07-21:
> bootstrap failed: please check /pi-gen/work/KioskPi/stage0/debootstrap.log

If you saw that line, then I don't see how the log could not be there: https://github.com/RPi-Distro/pi-gen/blob/2023-05-03-raspios-bullseye/scripts/common#L24

The presence of all those other log files seems to suggest the build may have gotten further?

Psychesnet, 2023-07-21:
Hi @XECDesign 

Thanks. I will check it later.