RPi-Distro/pi-gen #699: Running in chroot, ignoring command 'start'

cos-public, 2023-07-19:
I face with an issue similar to #657 . The error message is: `Running in chroot, ignoring command 'start'`

It seems that the problem is at the line:
`SUDO_USER="${FIRST_USER_NAME}" raspi-config nonint do_netconf 1` in `stage2/02-net-tweaks/01-run.sh`.  
When I chroot and try to execute it manually I get the same error. It seems the message comes from the line
`systemctl -q --no-block start "$ENABLE_SERVICE"` in `raspi-config`, `do_netconf()` function.

The `config` is pretty generic:
```
IMG_NAME="rpi"
RELEASE="bullseye"
LOCALE_DEFAULT="C.UTF-8"
TARGET_HOSTNAME="rpi"
KEYBOARD_KEYMAP="us"
KEYBOARD_LAYOUT="English (US)"
TIMEZONE_DEFAULT="Etc/UTC"
FIRST_USER_NAME="pi"
FIRST_USER_PASS="raspberry"
ENABLE_SSH=1
PUBKEY_SSH_FIRST_USER="<...>"
STAGE_LIST="stage0 stage1 stage2"
WPA_ESSID="<...>"
WPA_PASSWORD="<...>"
WPA_COUNTRY="UK"
```

When I remove `WPA_...` config lines the error goes away.

XECDesign, 2023-07-19:
That's not an error. Init systems, like systemd, don't start services when running in a chroot environment because it doesn't make sense to do so.

If you're having trouble setting the country, it's probably because UK is not a valid country code. I'm guessing you're after 'GB'.

cos-public, 2023-07-19:
Thank you! Yes, using `GB` fixes the issue.
Apparently `EU` from this article https://medium.com/@deltazero/making-kioskpi-custom-raspberry-pi-os-image-using-pi-gen-99aac2cd8cb6 is not a valid country code either. However both `UK` and `EU` are "reserved" country codes: https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2

It still boggers me that I can't see how the `systemctl -q --no-block start "$ENABLE_SERVICE"` in `raspi-config` doesn't get executed when the correct `WPA_COUNTRY` is set.

closed 2023-07-19 by XECDesign