RPi-Distro/pi-gen #703: Question: EOL software

andi34, 2023-08-04:
Hey!

On PiOS there's software in it's dependencies which reached EOL. Node.js v12 and PHP7 for example.

There's any chance the source lists get updated by default? I know, I can update those packages myself but since it's kind of a security issue I'd like to ask if there's a way to address that for everyone officially by default.

If it's the wrong place here to ask i'd be glad to know where to report it - also if this can be passed to the maintainer.

Best regards

Andi

XECDesign, 2023-08-04:
Although upstream maintainers for a bit of software may EOL it, that doesn't mean it's completely abandoned and isn't getting software updates. Packages in Raspberry Pi OS, in most cases, comes from Debian. Debian's package maintainers and security team look after such software, patching security issues as they come up. You can take a look at their FAQ [here](https://www.debian.org/security/faq).

Is there a particular CVE you're concerned about?

andi34, 2023-08-04:
Thanks for the information. No, there's no particular CVE I am concerned about yet. Besides the security concerns it's also hard for me to keep my OpenSource project users on a Raspberry Pi up to date without dealing to much with workarounds.

XECDesign, 2023-08-07:
Since Raspberry Pi OS is based on Debian stable, which is a fixed release distro, there's not much we can do.

On one hand, you miss out on new features, but the advantage is that your dependencies aren't constantly changing under you.

There have been many attempts to solve the issue with docker, electron, flatpak, appimage, snap, venvs and so on, but I feel like they cause more problems than they solve.

closed 2023-08-07 by XECDesign