RPi-Distro/pi-gen #704: Docker build fails with: unbound variable on darwin

half2me, 2023-08-31:
Steps to reproduce:
```
git clone --depth 1 https://github.com/RPI-Distro/pi-gen.git
cd pi-gen
echo "IMG_NAME=Foo" > config
./build-docker.sh
```

I get the following error: `./build-docker.sh: line 147: DOCKER_CMDLINE_PRE[@]: unbound variable`

Output of `docker --version`
```
Client:
 Cloud integration: v1.0.35-desktop+001
 Version:           24.0.5
 API version:       1.43
 Go version:        go1.20.6
 Git commit:        ced0996
 Built:             Fri Jul 21 20:32:30 2023
 OS/Arch:           darwin/arm64
 Context:           desktop-linux

Server: Docker Desktop 4.22.0 (117440)
 Engine:
  Version:          24.0
  API version:      1.43 (minimum version 1.12)
  Go version:       go1.20.6
  Git commit:       HEAD
  Built:            Sun Jul 30 02:09:41 2023
  OS/Arch:          linux/arm64
  Experimental:     false
 containerd:
  Version:          1.6.21
  GitCommit:        3dce8eb055cbb6872793272b4f20ed16117344f8
 runc:
  Version:          1.1.7
  GitCommit:        v1.1.7-0-g860f061
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0
```

reubenmiller, 2023-09-05:
This problem happens because MacOS has a very old version of bash by default which is missing

1. Install a newer version of bash (e.g. using homebrew). Afterwards make sure it is included in the `PATH` variable before the `/bin` folder.
2. Replace the shebang in the `build-docker.sh` script to the following:

    ```sh
    #!/usr/bin/env bash
    set -eu
    ```

Though you might run into some follow up problems (I'm currently doing the same thing and try to debug as I go).

reubenmiller, 2023-09-05:
FYI: I've created a PR to the `arm64` branch with a change to the `build-docker.sh` script to support running on MacOS with the default version of bash (e.g. version 3.x).

https://github.com/RPi-Distro/pi-gen/pull/705