RPi-Distro/pi-gen #710: unable to build bookworm version

jsiverskog, 2023-10-11:
hi,
i'm unable to build the arm64 version of bookworm. i'm running without docker on a rpi4.

it fails with errors such as:
```
Setting up raspi-firmware (1:1.20230829+ds-1~rpt3) ...
ln: failed to create symbolic link '/boot/overlays': Operation not permitted
dpkg: error processing package raspi-firmware (--configure):
 installed raspi-firmware package post-installation script subprocess returned error exit status 1
```
if i download the corresponding debian file for that package (from https://archive.raspberrypi.com/debian/pool/main/r/raspi-firmware/) i can see that it tries to do this:
```
# Ensure the target directory exists. See https://bugs.debian.org/887062
mkdir -p "${firmware_dst}/overlays"
if ! [ -e /boot/overlays ]; then
  ln -s firmware/overlays /boot/overlays
elif ! [ -L /boot/overlays ]; then
  echo "WARNING: raspi-firmware: /boot/overlays exists, but is not a symlink to ${firmware_dst}/overlays" >&2
fi
```

i suspect `${ROOTFS_DIR}` should be in there?

XECDesign, 2023-10-12:
Are you building from the latest commit? What filesystem are you building on?

jsiverskog, 2023-10-12:
ah, this only happens when building with `USE_QCOW2=1`. sorry, i forgot to mention my config.

is that way of building not supported anymore?

closed 2023-10-18 by jsiverskog