RPi-Distro/pi-gen #711: Tagging for 64-bit releases

jketterl, 2023-10-12:
Since the 64-bit releases of Raspberry Pi OS seem to be official by now, I would appreciate if you could provide git tags for the commit that was used to build the images that are available for download. Basically the same as with the release tags that are already available (for example the most recent one `2023-10-10-raspios-bookworm`) but taken from the arm64 branch.

I am using this repository to build custom Raspberry Pi images with preloaded components for our proejct. I would like to be able to provide arm64 based images as well.

I am aware that there is an `arm64` branch but I do prefer not to include potentially unstable changes. I have been merging the release tags from this repository and this has worked well so far, and I would like to apply the same practice for 64bit builds.

XECDesign, 2023-10-13:
Yeah, that makes sense. Added [2023-10-10-raspios-bookworm-arm64](https://github.com/RPi-Distro/pi-gen/releases/tag/2023-10-10-raspios-bookworm-arm64)

jketterl, 2023-10-13:
Thank you, much appreciated. First builds are done and working :)

closed 2023-10-13 by jketterl