RPi-Distro/pi-gen #713: doesn't have the component 'non-free-firmware'

builderjer, 2023-10-14:
I just started getting this error.
```
W: Skipping acquire of configured file 'non-free-firmware/i18n/Translation-en' as repository 'http://deb.debian.org/debian bullseye-updates InRelease' doesn't have the component 'non-free-firmware' (component misspelt in sources.list?)
E: Unable to locate package linux-image-rpi-v8
E: Unable to locate package linux-image-rpi-2712
E: Unable to locate package linux-headers-rpi-v8
E: Unable to locate package linux-headers-rpi-2712
```
I am using `https://github.com/usimd/pi-gen-action` to run pi-gen on github and has worked for hundreds of images, then just came up with this

XECDesign, 2023-10-16:
It looks like you're building a bullseye image from a bookworm branch. You'd need to use `bullseye-arm64` if that's what you're building.

builderjer, 2023-10-17:
Thank you, I did not look deep enough to see that there was a new branch

closed 2023-10-17 by builderjer