RPi-Distro/pi-gen #714: Bookworm Build Issues - RPI Zero 2 W

Ty9000, 2023-10-15:
Hello! Trying to build an extremely minimal RPI image - I create the image at the end of Stage 1; however, the RPI doesn't seem to like it very much - it shuts down after the initial boot and the journal complains about it not shutting down properly. I enabled UART to see its output. The only real thing I changed was to add some more packages in Stage 1

[Initial Boot.txt](https://github.com/RPi-Distro/pi-gen/files/12910181/Initial.Boot.txt)
[Second Boot.txt](https://github.com/RPi-Distro/pi-gen/files/12910182/Second.Boot.txt)

1stcall, 2023-10-15:
I believe you will need to build up to at least stage2 to install the raspberry pi firmware to the boot partition.

XECDesign, 2023-10-16:
Yeah, I wouldn't expect it to 'just work' and figuring out why would take a bit of digging. For general support for unsupported use cases like this, I think the forum is a better place to ask.

closed 2023-10-16 by XECDesign