RPi-Distro/pi-gen #722: export-image prerun fails

galla15, 2023-11-10:
I'm trying to build a custom lite image using the docker build script and I'm stuck on this error:

```bash
[12:19:43] End /pi-gen/stage2
[12:19:43] Begin /pi-gen/export-image
[12:19:43] Begin /pi-gen/export-image/prerun.sh
du: cannot access '/pi-gen/work/rpi-emulator'$'\r''/stage2/rootfs/proc/1547/task/1547/fd/3': No such file or directory
du: cannot access '/pi-gen/work/rpi-emulator'$'\r''/stage2/rootfs/proc/1547/task/1547/fdinfo/3': No such file or directory
du: cannot access '/pi-gen/work/rpi-emulator'$'\r''/stage2/rootfs/proc/1547/fd/4': No such file or directory
du: cannot access '/pi-gen/work/rpi-emulator'$'\r''/stage2/rootfs/proc/1547/fdinfo/4': No such file or directory
truncate: failed to truncate '/pi-gen/work/rpi-emulator'$'\r''/export-image/2023-11-10-rpi-emulator'$'\r''-lite.img' at 168898087354368 bytes: File too large
``` 

Any ideas?

j123b567, 2024-04-24:
Probably related to https://github.com/RPi-Distro/pi-gen/issues/537

XECDesign, 2024-04-30:
Is it resolved after f1c166a2833950a7c44fe19b01780723635a7aa3?