RPi-Distro/pi-gen #726: `raspbian.raspberrypi.com` vs `deb.debian.org` in `sources.list`

bradisbell, 2023-11-20:
In the arm64 branch of `stage0/00-configure-apt-files/sources.list`, `deb.debian.org` is referenced:

```
deb http://deb.debian.org/debian RELEASE main contrib non-free non-free-firmware
deb http://deb.debian.org/debian-security/ RELEASE-security main contrib non-free non-free-firmware
deb http://deb.debian.org/debian RELEASE-updates main contrib non-free non-free-firmware
# Uncomment deb-src lines below then 'apt-get update' to enable 'apt-get source'
#deb-src http://deb.debian.org/debian RELEASE main contrib non-free non-free-firmware
#deb-src http://deb.debian.org/debian-security/ RELEASE-security main contrib non-free non-free-firmware
#deb-src http://deb.debian.org/debian RELEASE-updates main contrib non-free non-free-firmware
```

In the master branch, it's `raspbian.reaspberrypi.com`:

```
deb [ arch=armhf ] http://raspbian.raspberrypi.com/raspbian/ RELEASE main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://raspbian.raspberrypi.com/raspbian/ RELEASE main contrib non-free rpi
```

Is there a reason for this difference?  Are armhf builds not available on `deb.debian.org`?

I ask because the performance of the Fastly-hosted `deb.debian.org` is much better than that of `raspbian.raspberrypi.com`.  We found that moving to `deb.debian.org` cut our Pi image building time about in half.  Perhaps the armhf users can make use of the Fastly-hosted Debian mirror as well, if possible.

XECDesign, 2023-11-22:
The difference is that one would build an image with a Raspbian base, suitable for armv6 pi models, while the other would use Debian and only work on armv7 and higher.

closed 2023-11-22 by XECDesign

bradisbell, 2023-11-22:
@XECDesign That's only due to this tag `[ arch=armhf ]` though, right?  Or is there something different about what's hosted on `raspbian.raspberrypi.com` beyond it just missing >=armv7 binaries?  Or, is it that `deb.debian.org` doesn't host `armv6` binaries?

XECDesign, 2023-11-22:
Debian's armhf is armv7 only. There's armel for armv6, but it's soft fp only.

You can read a bit about what Raspbian is and how it differs from Debian [here](http://www.raspbian.org/RaspbianAbout).