RPi-Distro/pi-gen #728: Using USE_QCOW2=1 fails

solskogen, 2023-11-25:
config:
```
IMG_NAME=Solskogen
USE_QCOW2=1

```
This is where is starts to fail

```
ln: failed to create symbolic link '/boot/overlays': Operation not permitted
run-parts: /etc/initramfs/post-update.d//z50-raspi-firmware exited with return code 1
run-parts: /etc/kernel/postinst.d/initramfs-tools exited with return code 1
dpkg: error processing package linux-image-6.1.0-rpi6-rpi-v8 (--configure):
 installed linux-image-6.1.0-rpi6-rpi-v8 package post-installation script subprocess returned error exit status 1
Setting up binutils-aarch64-linux-gnu (2.40-2) ...
Setting up binutils (2.40-2) ...
Setting up gcc-12 (12.2.0-14) ...
Setting up libgd3:arm64 (2.3.3-9) ...
Setting up linux-headers-6.1.0-rpi6-rpi-2712 (1:6.1.58-1+rpt2) ...
Setting up linux-headers-rpi-2712 (1:6.1.58-1+rpt2) ...
dpkg: dependency problems prevent configuration of linux-image-rpi-v8:
 linux-image-rpi-v8 depends on linux-image-6.1.0-rpi6-rpi-v8 (= 1:6.1.58-1+rpt2); however:
  Package linux-image-6.1.0-rpi6-rpi-v8 is not configured yet.

dpkg: error processing package linux-image-rpi-v8 (--configure):
 dependency problems - leaving unconfigured
Setting up linux-image-6.1.0-rpi6-rpi-2712 (1:6.1.58-1+rpt2) ...
I: /vmlinuz.old is now a symlink to boot/vmlinuz-6.1.0-rpi6-rpi-v8
I: /initrd.img.old is now a symlink to boot/initrd.img-6.1.0-rpi6-rpi-v8
I: /vmlinuz is now a symlink to boot/vmlinuz-6.1.0-rpi6-rpi-2712
I: /initrd.img is now a symlink to boot/initrd.img-6.1.0-rpi6-rpi-2712
/etc/kernel/postinst.d/initramfs-tools:
update-initramfs: Generating /boot/initrd.img-6.1.0-rpi6-rpi-2712
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8156b-2.fw for built-in driver r8152
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8156a-2.fw for built-in driver r8152
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8153c-1.fw for built-in driver r8152
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8153b-2.fw for built-in driver r8152
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8153a-4.fw for built-in driver r8152
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8153a-3.fw for built-in driver r8152
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8153a-2.fw for built-in driver r8152
ln: failed to create symbolic link '/boot/overlays': Operation not permitted
run-parts: /etc/initramfs/post-update.d//z50-raspi-firmware exited with return code 1
run-parts: /etc/kernel/postinst.d/initramfs-tools exited with return code 1
dpkg: error processing package linux-image-6.1.0-rpi6-rpi-2712 (--configure):
 installed linux-image-6.1.0-rpi6-rpi-2712 package post-installation script subprocess returned error exit status 1
Setting up libc-devtools (2.36-9+rpt2+deb12u3) ...
dpkg: dependency problems prevent configuration of linux-image-rpi-2712:
 linux-image-rpi-2712 depends on linux-image-6.1.0-rpi6-rpi-2712 (= 1:6.1.58-1+rpt2); however:
  Package linux-image-6.1.0-rpi6-rpi-2712 is not configured yet.
dpkg: error processing package linux-image-rpi-2712 (--configure):
 dependency problems - leaving unconfigured
Setting up linux-headers-6.1.0-rpi6-rpi-v8 (1:6.1.58-1+rpt2) ...
Setting up linux-headers-rpi-v8 (1:6.1.58-1+rpt2) ...
Processing triggers for libc-bin (2.36-9+rpt2+deb12u3) ...
Processing triggers for initramfs-tools (0.142) ...
ln: failed to create hard link '/boot/initrd.img-6.1.0-rpi6-rpi-v8.dpkg-bak' => '/boot/initrd.img-6.1.0-rpi6-rpi-v8': Operation not permitted
update-initramfs: Generating /boot/initrd.img-6.1.0-rpi6-rpi-v8
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8156b-2.fw for built-in driver r8152
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8156a-2.fw for built-in driver r8152
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8153c-1.fw for built-in driver r8152
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8153b-2.fw for built-in driver r8152
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8153a-4.fw for built-in driver r8152
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8153a-3.fw for built-in driver r8152
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8153a-2.fw for built-in driver r8152
ln: failed to create symbolic link '/boot/overlays': Operation not permitted
run-parts: /etc/initramfs/post-update.d//z50-raspi-firmware exited with return code 1
dpkg: error processing package initramfs-tools (--configure):
 installed initramfs-tools package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 raspi-firmware
 linux-image-6.1.0-rpi6-rpi-v8
 linux-image-rpi-v8
 linux-image-6.1.0-rpi6-rpi-2712
 linux-image-rpi-2712
 initramfs-tools
E: Sub-process /usr/bin/dpkg returned an error code (1)
[16:03:18] Unloading image
/root/pi-gen/work/Solskogen/stage0/rootfs: 10.5 GiB (11236798464 bytes) trimmed
/root/pi-gen/work/Solskogen/stage0/rootfs/tmp
/root/pi-gen/work/Solskogen/stage0/rootfs/sys
/root/pi-gen/work/Solskogen/stage0/rootfs/run
/root/pi-gen/work/Solskogen/stage0/rootfs/proc
/root/pi-gen/work/Solskogen/stage0/rootfs/dev/pts
/root/pi-gen/work/Solskogen/stage0/rootfs/dev
/root/pi-gen/work/Solskogen/stage0/rootfs/boot
/root/pi-gen/work/Solskogen/stage0/rootfs
/dev/nbd0 disconnected
```

j123b567, 2023-11-29:
See #648 and #710

XECDesign, 2024-02-23:
Closing, since this feature has been removed and will be replaced by a more reliable alternative.

closed 2024-02-23 by XECDesign

solskogen, 2024-02-23:
Great! Can you merge it into the arm64 branch as well, perhaps?