RPi-Distro/pi-gen #736: /boot/firmware changes missing in resize script

david-bla, 2023-12-14:
Hi pi-gen Team,

I am using pi-gen to create custom images that are deployed / should run without any interactive actions.

Since the change to bookworm and the new /boot/firmware directory this process was interrupted.

I am refering to the commit c856800c72c8a58e17d7ee20295605d479b00dff it seems like this changes are not compatible with the resize Script /usr/lib/raspi-config/init_resize.sh.

This Script is crashing because it try to get the Mount point of /boot and not /boot/firmware, so it thinks Boot is on loop device and root on sd device.

Even if I change /boot to /boot/firmware it does not run without interaction because it is asking yes/no question because the device is still mounted.

I am writing this on my Phone and was not able to find the sources of the resize Script to create a proper pull request. If needed I can create one lauter days.

Thank you verry much for your work 👍this is awesome. Best regards

XECDesign, 2023-12-14:
What's launching that script? It shouldn't be used anymore.

Resize is now handled in initramfs using [this script](https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/bookworm/initramfs-tools/scripts/local-premount/firstboot#L33)

david-bla, 2023-12-14:
@XECDesign thank you - that question was important for me.
I was about to write `it is added as init parameter to cmdline.txt` but I noticed that it was changed here -> [cmline patch](https://github.com/RPi-Distro/pi-gen/blob/2023-12-05-raspios-bookworm/stage2/01-sys-tweaks/00-patches/07-resize-init.diff)

for some reason this change was not merged properly into my pi-gen fork - I'll correct this and try to find the root cause of this issue (in my local fork).

So this issue is pointless now :stuck_out_tongue_winking_eye:, maybe it can serve as a reminder to remove the init_resize.sh at all.

Thank you

closed 2023-12-14 by david-bla