RPi-Distro/pi-gen #738: piwiz always starts

sagiadinos, 2023-12-22:
Hi,

The piwiz app start always in generated images. 

OS: Debian 12 

git clone --branch arm64 https://github.com/RPI-Distro/pi-gen.git

No changes in the stages
The process starts with this bash script:
--------------
export IMG_NAME="garlic-player"
export PI_GEN_RELEASE="Garlic-Player - SMIL based Digital Signage Player "
export BASE_DIR=/home/niko/pi-gen
export WORK_DIR=/home/niko/build_pi_image
export TARGET_HOSTNAME="garlic-player-rpi"
export ENABLE_SSH=1
export STAGE_LIST="stage0 stage1 stage2 stage3 stage4"

mkdir -p $WORK_DIR
cd /home/niko/pi-gen
bash ./build.sh
----------------------

The settings will be made correct, but after reboot, the wizard restarts again and again.

Do I set something wrong?

sagiadinos, 2023-12-29:
Enable ssh seams to be the problem.

remove 

`export ENABLE_SSH=1
`

and the exported image will start piwiz only once.

XECDesign, 2024-01-02:
I've just built an armhf image with `ENABLE_SSH=1` and piwiz behaved as expected.

Rebuilding an arm64 image, but I suspect it will behave the same way.

closed 2024-01-04 by XECDesign