RPi-Distro/pi-gen #739: build-docker.sh failes, error hidden and can't be accessed

zoff99, 2023-12-28:
default: I: Extracting util-linux...
    default: I: Extracting zlib1g...
    default: W: Failure trying to run: chroot "/pi-gen/work/Raspbian/stage0/rootfs" /bin/true
    default: W: See /pi-gen/work/Raspbian/stage0/rootfs/debootstrap/debootstrap.log for details
    default: rmdir: failed to remove '/pi-gen/work/Raspbian/stage0/rootfs/debootstrap': Directory not empty
    default: [21:09:08] bootstrap failed: please check /pi-gen/work/Raspbian/stage0/debootstrap.log

it fails like this. but the logfile is somewhere inside the container. i can't access it.

zoff99, 2023-12-31:
there seem to be 2 issues here:

1.  debootstrap.log ist not directly shown, but can not be reached since it's inside docker
i remember the content of that log being shown directly some months ago

2. an additional check for binfmt is missing
i needed to run `update-binfmts --enable` to have binfmt actually work

XECDesign, 2024-01-02:
Out of curiosity, what's the host distro?

zoff99, 2024-01-02:
vagrant vm ubuntu 22.04

zoff99, 2024-01-02:
my custom images build again with that fix:
https://github.com/zoff99/ToxBlinkenwall_raspi_lite_image/actions/runs/7376315433

XECDesign, 2024-01-02:
Hmm, a container running in VM running in a VM makes things a bit tricky to track down.

Since I don't use docker, I don't touch build-docker.sh. Support is mostly provided by other contributors.

However, what I think is happening is that because you're using `--no-install-recommends`, `binfmt-support` doesn't get installed, which would normally run `update-binfmts` for you.

zoff99, 2024-01-02:
its not using build-docker.sh now. it happens also with build.sh

XECDesign, 2024-01-02:
Just realised that `update-binfmts` contains `binfmt-support`. so if you're running it it means you must have the package installed.

Are you able to check the status/log of binfmt-support.service?

zoff99, 2024-01-02:
i think the first step should be instead of saying "here look in this file for the error messages" just do "cat thelogfile.log" to show the errors directly

XECDesign, 2024-01-02:
> yes the package is installed the kernel module loaded. but the formats are not registered. you can easily see it yourself:
> 
> ```
> git clone https://github.com/zoff99/ToxBlinkenwall_raspi_lite_image
> cd ToxBlinkenwall_raspi_lite_image/
> # remove the line with "update-binfmts --enable" from the script
> ./vagrant_build.sh
> ```

Seems to work for me with or without that line

XECDesign, 2024-01-02:
Also tried just bringing up ubuntu/jammy64 without any changes, cloning pi-gen, installing the dependencies and building an image. Also works without any need to load modules manually or mess with binfmt.

zoff99, 2024-01-02:
funky. now it works for me aswell. thats a bit scary. i could reproduce it many times. and even after rebooting my actually system it would break the same.

ok so lets focus on the first issue (if it breaks for some reason at the point).

```
    default: I: Extracting util-linux-extra...
    default: I: Extracting zlib1g...
    default: W: Failure trying to run: chroot "/home/vagrant/pi-gen/work/Raspbian/stage0/rootfs" /bin/true
    default: W: See /home/vagrant/pi-gen/work/Raspbian/stage0/rootfs/debootstrap/debootstrap.log for details
    default: rmdir: failed to remove '/home/vagrant/pi-gen/work/Raspbian/stage0/rootfs/debootstrap': Directory not empty
    default: [16:31:24] bootstrap failed: please check /home/vagrant/pi-gen/work/Raspbian/stage0/debootstrap.log
```

can it just cat the contents of the debootstrap.log instead? if thats easy to do

zoff99, 2024-01-02:
```
        if [ -d "$2/debootstrap" ] && ! rmdir "$2/debootstrap"; then
                cp "$2/debootstrap/debootstrap.log" "${STAGE_WORK_DIR}"
                log "bootstrap failed: please check ${STAGE_WORK_DIR}/debootstrap.log"
                return 1
        fi
```

do a "cat" instead

zoff99, 2024-01-02:
thanks for your help so far

XECDesign, 2024-01-04:
I don't think the log will tell you anything more interesting. The main error message already tells you that it's unable to run binaries in the chroot.

Whether the log file is just printed at the end or if the user needs to check it manually is not really a bug, but a matter of preference.