RPi-Distro/pi-gen #742: aplay: set_params:1352: Sample format non available

zoff99, 2024-01-06:
i flashed https://downloads.raspberrypi.com/raspios_lite_arm64/images/raspios_lite_arm64-2023-12-11/2023-12-11-raspios-bookworm-arm64-lite.img.xz to a new SD Card. and started it up on the a PI4.
its connected to a TV with HDMI0 port. picture shows up on TV but no audio playback is possible with alsa.

it always says: "aplay: set_params:1352: Sample format non available"

Available formats:
- IEC958_SUBFRAME_LE

how can i add all the missing formats?

if i take the sd card out of the PI4 and plug it into a PI5 i get all these:

Recognized sample formats are: S8 U8 S16_LE S16_BE U16_LE U16_BE S24_LE S24_BE U24_LE U24_BE S32_LE S32_BE U32_LE U32_BE FLOAT_LE FLOAT_BE FLOAT64_LE FLOAT64_BE IEC958_SUBFRAME_LE IEC958_SUBFRAME_BE MU_LAW A_LAW IMA_ADPCM MPEG GSM S20_LE S20_BE U20_LE U20_BE SPECIAL S24_3LE S24_3BE U24_3LE U24_3BE S20_3LE S20_3BE U20_3LE U20_3BE S18_3LE S18_3BE U18_3LE U18_3BE G723_24 G723_24_1B G723_40 G723_40_1B DSD_U8 DSD_U16_LE DSD_U32_LE DSD_U16_BE DSD_U32_BE

i don't know whats happening there.

```
pi@raspberrypi:~ $ aplay -D plughw:1,0 /usr/share/sounds/alsa/Front_Center.wav
Playing WAVE '/usr/share/sounds/alsa/Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
aplay: set_params:1352: Sample format non available
Available formats:
- IEC958_SUBFRAME_LE

pi@raspberrypi:~ $ speaker-test -c1 -D plughw:1,0

speaker-test 1.2.8

Playback device is plughw:1,0
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
Sample format not available for playback: Invalid argument
Setting of hwparams failed: Invalid argument

pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $ cat /proc/asound/card1/pcm0p/info 
card: 1
device: 0
subdevice: 0
stream: PLAYBACK
id: MAI PCM i2s-hifi-0
name: MAI PCM i2s-hifi-0
subname: subdevice #0
class: 0
subclass: 0
subdevices_count: 1
subdevices_avail: 1
pi@raspberrypi:~ $ cat /proc/asound/
card0/      card2/      devices     modules     pcm         timers      vc4hdmi1/
card1/      cards       Headphones/ oss/        seq/        vc4hdmi0/   version
pi@raspberrypi:~ $ cat /proc/asound/card
card0/ card1/ card2/ cards  
pi@raspberrypi:~ $ cat /proc/asound/cards 
 0 [Headphones     ]: bcm2835_headpho - bcm2835 Headphones
                      bcm2835 Headphones
 1 [vc4hdmi0       ]: vc4-hdmi - vc4-hdmi-0
                      vc4-hdmi-0
 2 [vc4hdmi1       ]: vc4-hdmi - vc4-hdmi-1
                      vc4-hdmi-1

```

zoff99, 2024-01-08:
with this https://downloads.raspberrypi.com/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/2021-05-07-raspios-buster-armhf-lite.zip it works on the PI4

XECDesign, 2024-01-08:
@popcornmix does this sound like a kernel issue?

zoff99, 2024-01-08:
same goes for the PI3 B+
so latest OS:
aplay via HDMI:

- PI5 -> work
- PI4 -> does not work
- PI3 B+ -> does not work

i am trying the full (instead of the lite) version now ...

zoff99, 2024-01-08:
funky with the full version: https://downloads.raspberrypi.com/raspios_full_arm64/images/raspios_full_arm64-2023-12-06/2023-12-05-raspios-bookworm-arm64-full.img.xz
it works on the PI4.

hmm. now how to figure out what is missing in the lite version so that it only work on the PI5

zoff99, 2024-01-08:
i think i am getting close.
if i change in `/usr/share/alsa/alsa.conf`

defaults.ctl.card 0
defaults.pcm.card 0

to

defaults.ctl.card 1
defaults.pcm.card 1

then aplay works. now why is the default not working?

zoff99, 2024-01-08:
ok installing this pkg: raspberrypi-ui-mods fixes the audio device order. but since that pulls in a zillion others like wayland and pulse i don't really know what happens.

i think alsa should work even on the lite images out of the box.

zoff99, 2024-01-08:
could be related to: https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/157

XECDesign, 2025-01-29:
Closing since it's not a pi-gen issue and I don't know enough about alsa. I suspect installing pipewire with the alsa module will work around the problem.

closed 2025-01-29 by XECDesign

CrabbyPete, 2025-01-29:
It required PulseAudio.

XECDesign, 2025-01-29:
Pulseaudio drags in a lot of dependencies. Pipewire is a lot lighter and is also what we use on the desktop.