RPi-Distro/pi-gen #745: Unabel to see /dev/ttyAMA0

RosiersRobin, 2024-01-11:
Hi,

I compiled the image and it worked out of the box, no biggy, however it seems I cannot see the `/dev/ttyAMA0` device. It's a RPI hat plugged in (qibixx). Anyone an idea? I normally can access it via minicom and run some serial command. Now, I cannot. /dev/TTYS0 doesn't do much either.

Any ideas here?

XECDesign, 2024-01-17:
I seem to have the same issue on my pi400, which only has tty[0-9]+ nodes.

What pi model are you using?

XECDesign, 2024-01-17:
@pelwell

pelwell, 2024-01-17:
Have you enabled the UART in the usual way (`enable_uart=1` in config.txt)?

Be careful with Bookworm - config.txt has moved to /boot/firmware/config.txt; there is a symbolic link from /boot/config.txt to /boot/firmware/config.txt, but it's quite easy (with the wrong editor) to end up overwriting the symbolic link with the new, modified config.txt and leaving /boot/firmware/config.txt untouched.

pelwell, 2024-01-17:
You will also not see /dev/ttyAMA0 (or whatever the Bluetooth UART is - ttyS0 if you're using the `miniuart-bt` overlay) because we are allowing the kernel to configure Bluetooth. The magic mechanism it is using effectively says "this is an HCI device, not a UART, so it shouldn't have a /dev/ttywhatever".

XECDesign, 2024-01-17:
Using `enable_uart=1` comes up with /dev/ttyS0 and then adding `dtoverlay=miniuart-bt` comes up with `/dev/ttyAMA0` as expected.

I'll close it since it's not a pi-gen issue either way.

@RosiersRobin, if you're still having problems after setting up `/boot/firmware/config.txt` appropriately, let us know.

closed 2024-01-17 by XECDesign

pelwell, 2024-01-17:
Confirmed working as expected on a Pi 400.

To explain a bit more, the UART is not enabled by default on most Bluetooth-equipped Pis. This is because UART0 (normally ttyAMA0, when not being an HCI device) is reserved for Bluetooth, leaving UART1 (ttyS0). Unfortunately the clock rate (and hence baudrate) of UART1 varies with the core clock, so to get a stable baudrate you need to stop the core clock changing, which either uses a bit more power than usual or reduces your performance, depending on whether you peg it high or low - neither of which is ideal if you don't care about the UART. That's why it's opt-in.