RPi-Distro/pi-gen #749: Customization Using RPi Imager Cause Images to Fail on Boot

dtcooper, 2024-01-30:
Hi there,

I created a "lite" image using the arm64 branch with config:

```bash
IMG_NAME=raspios-bookworm-arm64
STAGE_LIST='stage0 stage1 stage2'
```

The image builds fine, and when flashed with the Raspberry Pi Imager v1.8.5 (latest) it boots okay except I notice,

* The second (rootfs) partition has not grown. Does this imply that [`firstboot`](https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/bookworm/initramfs-tools/scripts/local-premount/firstboot) in [raspberrypi-sys-mods](https://github.com/RPi-Distro/raspberrypi-sys-mods) didn't run?

When I flash the image using any kind of customization (ie a changed user name), booting completely fails. Restarting results in the same failure. Upon inspecting my USB drive with the image on it afterwards, I notice the following:

1. The second (rootfs) partition has not grown. Same problem above?
2. Inspecting the journal for the boot on the rootfs, I get,
   ```
   ...
   Jan 29 19:58:24 raspberrypi (strun.sh)[518]: kernel-command-line.service: Failed at step EXEC spawning /boot/firstrun.sh: No such file or directory
   ...
   ```
3. I also note that `cmdline.txt` on the bootfs is as follows,
   ```
   console=serial0,115200 console=tty1 root=PARTUUID=8e40417c-02 rootfstype=ext4 fsck.repair=yes rootwait systemd.run=/boot/firstrun.sh systemd.run_success_action=reboot systemd.unit=kernel-command-line.target
   ```
   The relevant part being `systemd.run=/boot/firstrun.sh` instead of `/boot/firmware/firstrun.sh`. Does this imply that [`imager_fixup`](https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/bookworm/initramfs-tools/scripts/local-bottom/imager_fixup) in [raspberrypi-sys-mods](https://github.com/RPi-Distro/raspberrypi-sys-mods) didn't run?

Thanks!

XECDesign, 2024-01-30:
Just flashed last night's arm64 lite image using rpi-imager 1.8.5 with customisations applied. Then instead of booting normally, my pi400 just powers off when it gets to a certain point. Looking into it now, thanks.

XECDesign, 2024-01-30:
Oh, looks like initramfs is missing altogether from /boot/firmware.

dtcooper, 2024-01-30:
Thanks for looking into this. Out of curiosity, where can I download the CI nightly images? Thanks again!

XECDesign, 2024-01-30:
Okay, someone requested that we don't have a hard dependency on initramfs-tools, so I moved it to recommends. So now, there's no guarantee that initramfs-tools is set up before the kernel. In this case, the initramfs kernel postinst hooks are still named .dpkg-new at the time the kernel postinst script runs run-parts. So, the hook that would normally create initramfs doesn't run.

A few options here:

1) Revert the dependency change
1) Explicitly install initramfs-tools before the kernel
1) Run `update-initramfs -k all -c` when the image is exported.

I think I'll go for the third option for now and then if we find other people are getting burned by it, I'll revert the change altogether.

> Thanks for looking into this. Out of curiosity, where can I download the CI nightly images? Thanks again!

Thanks for catching the issue! The nightly images are not publicly available, they're just used internally for testing changes before they are shipped out.

closed 2024-01-30 by XECDesign

dtcooper, 2024-02-03:
Btw, confirming your fix addresses the issue perfect. Thanks again!

XECDesign, 2024-02-03:
Thanks for checking

rviollette, 2024-02-13:
Hi there,
I confirm this comit c20ce09facaaadcc480a7bbda4fe100fcacdee12 fixes the missing initramfs issue. 
Firstboot script gets called, the partition is extended as expected, etc.

Thanks for the report & fix

commit c20ce09f (2024-01-30): export-image: make sure initramfs is created

Fixes #749