RPi-Distro/pi-gen #751: Use gpt and btrfs as root filesystem

sithmein, 2024-02-10:
It should be possible to use btrfs instead of ext4 as the root file system. There are several instruction on the internet how an existing installation can be converted but they are complicated and error-prone. Therefore having an image that uses btrfs in the first place would be quite useful.
In addition using storage devices with more then 2TB (the Raspberry Pi 5's new PCIe interface makes this possible easily) is currently not possible since MBR does not support partitions with more than 2TB. Therefore changing to gpt is required.

XECDesign, 2024-02-13:
GPT is not officially supported by the bootloader and Raspberry Pi OS currently only supports ext4 or nfs.

We may come back to btrfs at some point in the future, but that's not currently on the task list.

closed 2024-02-13 by XECDesign

sithmein, 2024-02-16:
What exactly do you mean with _only supports ext4 or nfs_? btrfs clearly works on Raspberry Pi OS.