RPi-Distro/pi-gen #758: When building an image with zfs-dkms with DEBIAN_FRONTEND=noninteractive still getting confirmation window

VeloAddict, 2024-02-28:
In stage 2 00-packages I added : `zfs-dkms zfsutils-linux`.
In config I added `DEBIAN_FRONTEND=noninteractive`
Started a build. I was not expecting this:
[image]

What can I do to now have to interact with that prompt?

VeloAddict, 2024-02-28:
I also tried running this in on_chroot:
`DEBIAN_FRONTEND=noninteractive apt install zfs-dkms zfsutils-linux`

But got the same result.

VeloAddict, 2024-02-28:
I figured it out.
In `00-debconf` I added:
`zfs-dkms zfs-dkms/note-incompatible-licenses note true`

Maybe this will be helpful for others.

VeloAddict, 2024-02-28:
I figured it out.
In 00-debconf I added:
zfs-dkms zfs-dkms/note-incompatible-licenses note true

Maybe this will be helpful for others.

closed 2024-02-28 by VeloAddict

XECDesign, 2024-03-04:
Thanks for posting the solution for others