RPi-Distro/pi-gen #759: Custom raspberry OS image cannot boot up when boot partition size is 64MB or 128MB

vanbuong, 2024-03-06:
Hello, I tried to build the custom image with a smaller size than the current Raspberry OS Lite image for the CM3 4GB eMMC module.
In /export-image/prerun.sh, I change the size of the boot partition to 64MB or 128MB

BOOT_SIZE="$((**64 * 1024 * 1024**))"
ROOT_SIZE=$(du --apparent-size -s "${EXPORT_ROOTFS_DIR}" --exclude var/cache/apt/archives --exclude boot/firmware --block-size=1 | cut -f 1)

In /stage0/02-firmware/01-packages, I keeps only these package: raspi-firmware, linux-image-rpi-v7l

The build process was completed successfully, but after flashing, the CM3 module cannot boot up, and the green LED stays on.

How can I resolve this issue? Please give me some advice. Thanks

XECDesign, 2024-03-06:
Did it work before you made that change? Are those the only two changes you made?

Off the top of my head, I can't think of why it would cause issues. Do the checksums of the files on the boot partition match what you'd expect?

If that's all it is, I can try to reproduce the problem a bit later and take a closer look.

XECDesign, 2024-03-11:
Reproduced the issues. Taking a look now.

XECDesign, 2024-03-11:
Okay, the issue is that you need either v7l or v8 kernel, not v7.

XECDesign, 2024-03-11:
Seems like I'm a bit too trigger-happy to close this one. Nope, that didn't fix it for me either. Starting to look like a bootloader issue.

XECDesign, 2024-03-11:
@timg236 It looks like there's an issue with booting from a 64MB FAT32-formatted partition. File contents seem identical between working and non-working cases, but it doesn't even get to the splash screen. Enabling UART output in bootcode.bin didn't seem to work either.

I'm not quite sure what the limitations are, but in the working case I see:

```
1044480 bytes per FAT (= 2040 sectors)
Data area starts at byte 2105344 (sector 4112)
    261115 data clusters (534763520 bytes)
63 sectors/track, 128 heads
```

In the non-working case, it's:

```
Warning: Filesystem is FAT32 according to fat_length and fat32_length fields,
  but has only 32624 clusters, less than the required minimum of 65525.
  This may lead to problems on some systems.

    131072 bytes per FAT (= 256 sectors)
Data area starts at byte 278528 (sector 544)
     32624 data clusters (66813952 bytes)
63 sectors/track, 64 heads
```

I'm guessing the issue is that we force `-F 32 -s 4` options, when they're not appropriate for a smaller partition size, but I have no idea.

I've saved a backup of the non-working partition for when you get back, in case it's useful (pi-gen/bootpart.img.xz).

vanbuong, 2024-03-13:
@XECDesign Thank you so much, I tried your options then RPi can boot up.

XECDesign, 2024-03-14:
Thanks for checking. Hopefully [this commit ](https://github.com/RPi-Distro/pi-gen/commit/3b5e214f5ec4e23323bdd489b999536c487fcd12) means that nobody else will hit this issue.

closed 2024-03-14 by XECDesign