RPi-Distro/pi-gen #760: MacOS + Docker: qemu-user-static check from #685 breaking build early

MauAbata, 2024-03-07:
I ran into an issue setting up pi-gen on MacOS Monterey 12.6 where qemu-user-static wasn't found, and it prompted me to install that package. I couldn't find any Mac equivalent, and after some digging discovered a comment on the original PR that references this issue:

              This PR broke my build-docker.sh command which I've been using for years on MacOS. The Shell script exits with the error message `qemu-arm-static not found (please install qemu-user-static)` although the build is working fine the versions before the PR. I am using MacOS Sonoma Version 14.0 (23A344). I have not found any way to install the missing package so I moved back in time using a few tags before to `git checkout 2023-02-21-raspios-bullseye`.

_Originally posted by @JavanXD in https://github.com/RPi-Distro/pi-gen/issues/685#issuecomment-1860931266_
            

Rolling back to `2023-02-21-raspios-bullseye` was able to generate a successful build, so I wonder if that qemu check isn't necessary on MacOS. I am not confident enough about Mac environments to be able to say for sure, so perhaps we can get some insight here.

JavanXD, 2024-03-07:
> to prevent possible problems on various distros @moh53n 

The PR broke the MacOS build, could you have a look please?

I'm still interested in getting the build process working on my MacBook again.

moh53n, 2024-03-08:
One way is to just ignore the whole process on non-linux environments, but there might be a better solution.
I don't have a Mac system and I'm unfamiliar with its OS.
Do you need to manually handle the qemu registration? How was the build process before this commit?

eins78, 2024-04-22:
@MauAbata @JavanXD Are you using an Intel or Apple Silicon Mac? I do not get this error and since `uname -m` returns `arm64` on Apple Silicon, the code in the PR you've linked should not on Apple Silicon/ARM processors.

If the build did indeed work on Intel Macs before the PR #685, then a quick fix could be to check for macOS and then skip the binfmt registration, like:
`uname -o | grep Darwin && binfmt_misc_required=0`

JavanXD, 2024-04-22:
> Are you using an Intel or Apple Silicon Mac? 
Intel, but I just switched to M3, so I can try it soon.

filthsystem, 2024-05-01:
> `uname -o | grep Darwin && binfmt_misc_required=0`

Tried this on an x86 Mac, `uname -s` is what you're looking for but otherwise it fixed the issue for me!