RPi-Distro/pi-gen #761: /usr/bin/raspi-config: 847: lxpanelctl: not found

ampledata, 2024-03-13:
It looks like the addition of the call to lxpanelctl at https://github.com/RPi-Distro/raspi-config/commit/dd6647646a71c14664c951c59cc2fb27b8a55162 is causing pi-gen to fail at stage2/02-net-tweaks/01-run.sh

```
[23:59:17] End /home/gba/src/SNS/aryaos/pi-gen/stage2/02-net-tweaks/00-packages
[23:59:17] Begin /home/gba/src/SNS/aryaos/pi-gen/stage2/02-net-tweaks/01-run.sh
'files/wpa_supplicant.conf' -> '/home/gba/src/SNS/aryaos/pi-gen/work/aryaos-1.0.0-beta1/stage2/rootfs/etc/wpa_supplicant/wpa_supplicant.conf'
/usr/bin/raspi-config: 847: lxpanelctl: not found
make: *** [Makefile:18: build] Error 127
```

My pi-gen config:

```
IMG_NAME="aryaos-1.0.0-beta1"
PI_GEN_RELEASE="AryaOS"
TARGET_HOSTNAME="aryaos"
FIRST_USER_NAME="pi"
FIRST_USER_PASS="example"
STAGE_LIST="stage0 stage1 stage2 stage3-wifi"
DISABLE_FIRST_BOOT_USER_RENAME=1
RELEASE="bookworm"
WPA_COUNTRY="US"
LOCALE_DEFAULT="en_US.UTF-8"
KEYBOARD_LAYOUT="English (US)"
KEYBOARD_KEYMAP="us"
ENABLE_SSH=1
```

pi-gen is arm64@6fb311e7fb57b4e7980ae3270c25b569b19dfe8f

XECDesign, 2024-03-13:
So what's making the `if pgrep lxpanel` check pass? Is it that you're running in on PiOS and since we bind /proc, it sees the host's lxpanel?

I guess raspi-config should additionally check whether it's running in a chroot environment in that case. But at this point, that's just a guess, since it's not failing to build in our CI system.

ampledata, 2024-03-13:
@XECDesign Where is the `if pgrep lxpanel` check located?

I'm building on a Debian 12 host:
`Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
`

`Linux kelp 5.10.0-26-amd64 #1 SMP Debian 5.10.197-1 (2023-09-29) x86_64 GNU/Linux`

ampledata, 2024-03-13:
Aye, my host is running LXDE.

XECDesign, 2024-03-13:
Yeah, that would be it then. I'll update raspi-config.

XECDesign, 2024-03-13:
If you do a clean build starting from stage0, I'm hoping it should work now.

ampledata, 2024-03-13:
Looks good!

closed 2024-03-13 by ampledata