RPi-Distro/pi-gen #766: adduser error when setting FIRST_USER_NAME in config

Siecje, 2024-04-25:
I'm using `FIRST_USER_NAME` and `DISABLE_FIRST_BOOT_USER_RENAME=1` to ensure a user is created.

Should I keep the default and create the new user in one of the stages?

`sudo ./build.sh`

```
[15:27:55] Begin /home/giatec/pi-gen/stage2/01-sys-tweaks/01-run.sh
hwclock.sh.service is not a native service, redirecting to systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install disable hwclock.sh
Synchronizing state of ssh.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable ssh
resize2fs_once.service is not a native service, redirecting to systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable resize2fs_once
adduser: The user `foo' does not exist.
```

It seems to be this line.
https://github.com/RPi-Distro/pi-gen/blob/165e9602376b1ee25d8599389e5ae5643dd9754c/stage2/01-sys-tweaks/01-run.sh#L53

config
```
IMG_NAME="baz"
RELEASE="bookworm"
DEPLOY_COMPRESSION="xz"
TARGET_HOSTNAME="baz"
LOCALE_DEFAULT="en_US.UTF-8"
KEYBOARD_KEYMAP="us"
KEYBOARD_LAYOUT="English (US)"
TIMEZONE_DEFAULT="UTC"
FIRST_USER_NAME="foo"
FIRST_USER_PASS="bar"
DISABLE_FIRST_BOOT_USER_RENAME=1
WPA_COUNTRY="US"
ENABLE_SSH="1"
STAGE_LIST="stage*"
```

I have removed some of the default packages. These are the packages that I've kept.
```shell
find . -type f -regex '.*[0-9]+-packages.*' -exec cat {} \; 2>/dev/null
gcc
python3-dev
wpasupplicant
firmware-atheros
firmware-brcm80211
firmware-libertas
firmware-misc-nonfree
firmware-realtek
raspberrypi-net-mods
network-manager
net-tools
ssh
less
fbset
sudo
psmisc
console-setup
keyboard-configuration
gpiod
python3-libgpiod
python3-gpiozero
pigpio
python3-pigpio
raspi-gpio
python3-rpi.gpio
python3-spidev
python3-smbus2
avahi-daemon
ca-certificates
curl
fake-hwclock
usbutils
dosfstools
raspberrypi-sys-mods
pi-bluetooth
htop
ssh-import-id
unzip
zip
file
python3-venv
raspberrypi-archive-keyring
initramfs-tools
raspi-firmware
linux-image-rpi-v8
linux-headers-rpi-v8
locales
systemd-timesyncd
netbase
userconf-pi
```

Siecje, 2024-04-26:
I think this issue was due to caching the build.

How do you rebuild everything?

I'm not able to remove everything from `work/`.

To work around this I am been cloning the directory

```shell
cd ..
git clone pi-gen/ pi2
# then later
git clone pi2/ pi3
```

closed 2024-04-26 by Siecje

XECDesign, 2024-04-29:
To do a clean build, run it with CLEAN=1.

Trying to delete work after a failed build is going to end badly. It would still have mountpoints from the chroot steps (/proc, /sys, /dev/ and so on), so you end deleting things you need for your host OS to work. If you've tried to do that, you should reboot.

To do it properly, unmount everything under the work directory first, check if you have any .img files attached to any loopback devices and detach those.

At some point, I need to remove this particular foot shotgun from pi-gen and make sure that it leaves the system free of extra mountpoints and loopback devices.