RPi-Distro/pi-gen #769: Change docker to build in the host filesystem directly

pevsonic, 2024-04-28:
Hi all,

I've been using build-docker.sh (which is handy, thanks!) but it seems unusual that the work / deploy exist within the container which is somewhat counter-intuitive / involved to work with especially when doing incremental builds. Is by design intent or is it just where things are? 

I've got some build environments I put together for building yocto / ptxdist / buildroot on old distros that I forked off from the great work done by the CROPS project where the build container is used solely to contain all build tools and operates directly on the source tree present on the host filesystem as the user which is really helpful. 

Would it be of interest if I made a fork of pi-gen to integrate this mechanism? I've done a quick proof of concept fork and it seems to work well - looking to the user as if its a simple wrapper to "build.sh" and produces work/ & deploy/ just as if you ran build.sh natively. Is there a standard set of regression tests you use? Im imagining at a least builds of recent images on linux, osx and windows?

XECDesign, 2024-04-29:
I don't use or look after the docker files (*). If you send a PR, it looks clean, doesn't add any unnecessary complexity, pull in containers from "some guy" and other Docker users are happy with it, I'll merge it.

* The official images are built in docker, but not using the docker files from this repo.

pevsonic, 2024-05-08:
Thanks! Ill close this for now and create a PR if needed - now Ive fixed the non-docker build my end making docker changes isnt such a priority for me at this time...