RPi-Distro/pi-gen #775: Add new one partition to image

pompushko, 2024-05-08:
Hello.

How can I add a new small partition like 64mb during build and store some data on it?
I just want to make a read-only image and I need some user-space partition.

Thank you!

XECDesign, 2024-05-08:
That's not currently a built-in feature, but we can treat it as a feature request for later.

In the meantime, you could edit `export-image/prerun.sh` to do it.

There may be other parts of Raspberry Pi OS which would be surprised to see an extra partition and might not know what to do with it. For example, I'd expect the [resize script](https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/bookworm/initramfs-tools/scripts/local-premount/firstboot) to fall over. However, I'm guessing that's not really a concern with a read-only rootfs.

nullromo, 2025-02-18:
I am currently trying to set up [SWUpdate](https://sbabic.github.io/swupdate/) on my Raspberry Pi, and I want to use the [double copy with fallback](https://sbabic.github.io/swupdate/overview.html#double-copy-with-fall-back) method. This means I need to have 2 partitions that I can boot from. Ideally I would be able to have 4 total partitions

- `bootfs` for the bootloader
- `rootfs-1` for copy 1
- `rootfs-2` for copy 2
- `data` for shared data that can persist across software updates

However, I'm struggling to figure out how to get pi-gen to build an image that has more partitions.

All that to say I would also be interested in this kind of feature, and I'm sure others in my situation would be as well.

XECDesign, 2025-02-25:
https://github.com/raspberrypi/rpi-image-gen may be worth taking a look at, as it seems more appropriate for that use case.

pi-gen is primarily a tool to generate the official Raspberry Pi OS image and some lightly modified derivatives. rpi-image-gen is for more embedded and industrial use cases.

nullromo, 2025-02-25:
Thank you for the pointer. I will take a look and see.

nullromo, 2025-02-25:
I was able to use `rpi-image-gen` to build an image, and I set `device_ssh_user1=y` in order to actually be able to connect to it. 

However, my existing CI pipeline runs on a build server in the cloud, not on a raspberry pi. When I tried to run `rpi-image-gen` on Ubuntu under WSL, I got the error `E: arm64 can neither be executed natively nor via qemu user emulation with binfmt_misc`. Pi-gen runs under Docker on WSL and on my build server just fine, so if building from a physical raspberry pi is a requirement for `rpi-image-gen`, then it won't work as well in my case.

learmj, 2025-02-26:
Hi

> if building from a physical raspberry pi is a requirement for rpi-image-gen, then it won't work as well in my case.

I do plan on investigating whether rpi-image-gen can run under qemu (like pi-gen). I haven't done so yet. However, I think that under docker there may be quirks as a result of using unshare (e.g. I'm not 100% sure if it'd be possible to bind mount /proc from the container into the chroot using unshare).
If this is something you need, please raise a ticket under rpi-image-gen with all the details, plus any failing traces, and I'll take a look when I can. The error cited is from mmdebstrap but I'd have expected a line or two before that one.

nullromo, 2025-03-04:
Actually I think running on a pi will be fine for me, as long as I can set up a dedicated pi with gitlab-runner and trigger builds on there. I tried doing this, but ran into [this issue](https://github.com/raspberrypi/rpi-image-gen/issues/26).

learmj, 2025-03-04:
Have a look at the examples in ```rpi-image-gen```. Seems to me this ticket belongs there rather than here.

XECDesign, 2025-03-05:
Since there are no plans for this in pi-gen any time soon, I'll close as not planned. Hopefully `rpi-image-gen` can cover the use case you're after.

closed 2025-03-05 by XECDesign