RPi-Distro/pi-gen #778: Using a custom configuration, RPi never leaves setup

greenpdx, 2024-06-15:
Using a custom config, after setup, it reboots into setup again and again.

Also the image is not gzipped.
 

IMG_NAME='Computadorita-arm64'
PI_GEN_RELEASE='dorita01'
DEPLOY_COMPRESSION='gz'
LOCALE_DEFAULT='es_CR.UTF-8'
TARGET_HOSTNAME='computadorita'
KEYBOARD_KEYMAP='latam'
KEYBOARD_LAYOUT='Spanish (CR)'
TIMEZONE_DEFAULT='America/Costa_Rica'
FIRST_USER_NAME='dorita'
WPA_COUNTRY='CR'
ENABLE_SSH=1

greenpdx, 2024-06-16:
If FIRST_USER_NAME is set the booting process fails.

XECDesign, 2024-06-24:
Just so I understand the desired behaviour a bit better,  what's the purpose of setting `FIRST_USER_NAME` when it will be renamed on first boot?

XECDesign, 2024-06-24:
The root of the issue seems to be that https://github.com/raspberrypi-ui/piwiz has `pi` hard-coded in a few places. The issue doesn't happen on the lite image.

But I'm still not sure I understand why this is a config somebody would try. Although the issue will be fixed in piwiz, it may be worth changing pi-gen to detect this pre-flight and error out early, letting the user know the config is not valid. Unless, there's a use case where it makes sense that I'm missing.

greenpdx, 2024-06-24:
I want to use RPi in an enterprise environment. Where it boots and
configures itself.  I am also getting SELinux running on it.

XECDesign, 2024-06-25:
In other words for you the issue isn't that the wizard runs multiple times, it's that it runs at all?

In that case, you'd need to set `DISABLE_FIRST_BOOT_USER_RENAME=1` and `FIRST_USER_PASS` to something. There's a bit more information in the README.

XECDesign, 2024-06-25:
Closing since the piwiz issue has been forwarded on and will be fixed after the next image release and the rest of the behaviour is as described in the README.

closed 2024-06-25 by XECDesign