RPi-Distro/pi-gen #783: resize2fs_once service no longer resizes root fs.

theGOTOguy, 2024-07-27:
I'm not sure when this got broken, but it did correctly work at 9249e146d2e3987adf2ea0e5e19e295d37f3f886 and does not work at dca93d24c7dfb943d84601d2bc4a089f9b66b87f.

Service actually runs, but doesn't detect that a resize needs to happen.  

[image]

The file system didn't get resized:

[image]

Running `raspi-config --expand-rootfs` does correctly resize the root file system, though.

XECDesign, 2024-07-29:
It looks like the partition table isn't getting adjusted on first boot in your case.

Is your image skipping [firstboot](https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/bookworm/usr/lib/raspberrypi-sys-mods/firstboot) in some way?

theGOTOguy, 2024-07-29:
I don't believe so:  you can see from the screenshot that resize2fs_once actually ran.  It just didn't do anything.

I did try manually running resize2fs from the CLI, and it said "nothing to do," even though it in fact had plenty to do.  

Running `raspi-config --expand-rootfs` did correctly resize the root fs.

XECDesign, 2024-07-29:
Note that firstboot is not resize2fs_once. I saw that resize2fs ran, but that's exactly how it would behave if firstboot didn't.

Think of the partition as a container and the filesystem as the content. firstboot edits the partition table to expand the partition table. Then resize2fs_once expands the filesystem to fill the partition. If the partition has not been expanded, there is nowhere for the filesystem to expand into. You can use `sudo fdisk -l /dev/mmcblk0` to confirm this.

theGOTOguy, 2024-07-29:
Oh, that's a good point.  I will have to produce a new image to confirm.

theGOTOguy, 2024-07-31:
Apologies, I produced a vanilla Raspbian Lite build from head and was unable to reproduce this issue.

It must be something specific to my custom build causing firstboot to fail.

Closing unable to reproduce.

closed 2024-07-31 by theGOTOguy