RPi-Distro/pi-gen #785: Improvement:  Build Default Image using a GitHub Action

theGOTOguy, 2024-07-28:
For my own custom build utilizing Pi-Gen, I have written [a GitHub action](https://github.com/simpleairquality/SimpleAQ-Device/blob/main/.github/workflows/build.yml) that automatically builds the image, which is then available to download directly through GitHub.

This would:

1. Automatically confirm on each push that the build process isn't broken.
2. Provide users with a ready-built Raspbian image with the most common settings for download and testing.
3. Serve as a blueprint for others who might like to build custom images based on Raspbian.

Please feel free to use my existing, open-source action above at a starting point.  Or, if there is interest from the devs, I will be happy to contribute a PR for you that would build a Raspbian image with the most common settings.

XECDesign, 2024-07-29:
That's great, thanks!

We use Gitlab CI for our official builds, but having a github workflow as an example for others would be useful.

I'd merge a PR that builds the default images without customisation.

I'm guessing Github doesn't provide enough disk space to build non-lite images, but that problem should be solved when I add COW support for stage copies.

jedahan, 2024-09-29:
Here's [another example workflow](https://github.com/alexnathanson/solar-protocol/blob/main/.github/workflows/main.yml) and the [build-image.sh](https://github.com/alexnathanson/solar-protocol/blob/main/utilities/build/build-image.sh) that runs inside the build step. Couldn't have gotten this far without @theGOTOguy work, so wanted to reciprocate.

SvenFinn, 2025-02-27:
Just to add my two cents:

I've run pi-gen builds on Github Actions for a project before, running into - seemingly - the same limitations as you regarding github actions storage.
During my project we modified the copy_previous function to delete the work directory of the previous build step after copying it, to reduce the amount  of (unnecessary) space taken up by older stages.

This has two major drawbacks:

- The EXPORT_IMAGE step is ran at the end of the build, using the work directories of each step that is to be exported. Deleting the work directory of a step makes it impossible to export this step. Therefore, when using this space optimization, you can only export the last build stage to an image.
- If the build fails, you cannot reuse the state, as the work directories of all previous stages are missing. When building using GitHub Actions, I don't think this is an issue.

To avoid the first drawback, you could restructure pi-gen to export images after it finishes building the stage, but that would require restructuring pi-gen.