RPi-Distro/pi-gen #788: bookworm build fails as of 08-05-2024: GPG error

daliang20, 2024-08-05:
I'm running pi-gen in a docker-build trying to build bookworm.

Last week, I was able to generate a lite image (stages 3-5 skipped); however, as of August 5, 2024 I am unable to get past stage 0.

This is a snippet of last weeks build for reference (gets past stage 0):
```
[17:24:09] Begin /pi-gen/stage0/00-configure-apt/00-run.sh
Get:1 http://archive.raspberrypi.com/debian bookworm InRelease [39.0 kB]
Hit:2 http://raspbian.raspberrypi.com/raspbian bookworm InRelease
Get:3 http://raspbian.raspberrypi.com/raspbian bookworm/rpi armhf Packages [1360 B]
Get:4 http://archive.raspberrypi.com/debian bookworm/main armhf Packages [427 kB]
Get:5 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages [426 kB]
Fetched 893 kB in 9s (103 kB/s)
Reading package lists...
W: http://raspbian.raspberrypi.com/raspbian/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Reading package lists...
Building dependency tree...
Calculating upgrade...
The following packages will be upgraded:
  dpkg libc-bin libc6 libpam-modules libpam-modules-bin libpam-runtime
  libpam0g libssl3 openssl
9 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 7667 kB of archives.
After this operation, 53.2 kB disk space will be freed.
```

Getting an issue specifically after today that won't go past stage 0:

```
I: Configuring ca-certificates...
I: Base system installed successfully.
[22:35:30] End /pi-gen/stage0/prerun.sh
[22:35:30] Begin /pi-gen/stage0/00-configure-apt
[22:35:30] Begin /pi-gen/stage0/00-configure-apt/00-run.sh
Get:1 http://archive.raspberrypi.com/debian bookworm InRelease [39.0 kB]
Hit:2 http://raspbian.raspberrypi.com/raspbian bookworm InRelease
Err:1 http://archive.raspberrypi.com/debian bookworm InRelease
  At least one invalid signature was encountered.
Err:2 http://raspbian.raspberrypi.com/raspbian bookworm InRelease
  At least one invalid signature was encountered.
Reading package lists...
W: GPG error: http://archive.raspberrypi.com/debian bookworm InRelease: At least one invalid signature was encountered.
E: The repository 'http://archive.raspberrypi.com/debian bookworm InRelease' is not signed.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://raspbian.raspberrypi.com/raspbian bookworm InRelease: At least one invalid signature was encountered.
[22:35:40] Build failed
real	6m23.235s
user	0m0.079s
sys	0m0.060s
Cleaning up project directory and file based variables
```

A [stackoverflow post](https://stackoverflow.com/questions/62473932/at-least-one-invalid-signature-was-encountered) suggests that the docker build cache was full; I pruned that and tried again - same issue.

I haven't tried a bulleyes build yet.

daliang20, 2024-08-05:
I just tried this on a local machine using `build-docker.sh` and it seemed to worked. There must something something wrong with my gitlab-ci

closed 2024-08-05 by daliang20

XECDesign, 2024-08-06:
This is related to the version of qemu on the host.

https://gitlab.com/qemu-project/qemu/-/issues/633

Whatever distro your gitlab runner is on is propably quite old.