RPi-Distro/pi-gen #794: keyboard settings are not working

CvH, 2024-09-23:
It looks like the keyboard settings are properly set during image creation but are not actually used from wayfire.

`debconf-show keyboard-configuration` shows me the proper configuration but the settings are missing at the `wayfire.ini` .

my pi-gen config
```
KEYBOARD_KEYMAP=de
KEYBOARD_LAYOUT=Deutsch
FIRST_USER_NAME=admin
FIRST_USER_PASS=1234
DISABLE_FIRST_BOOT_USER_RENAME=1
```
`debconf-show keyboard-configuration`
```
* keyboard-configuration/xkb-keymap: de
  keyboard-configuration/modelcode: pc105
  keyboard-configuration/toggle: No toggling
* keyboard-configuration/optionscode:
  keyboard-configuration/switch: No temporary switch
  keyboard-configuration/store_defaults_in_debconf_db: true
* keyboard-configuration/compose: No compose key
  keyboard-configuration/unsupported_config_options: true
  keyboard-configuration/unsupported_options: true
  keyboard-configuration/layout:
  keyboard-configuration/unsupported_config_layout: true
* keyboard-configuration/model: Generische PC-Tastatur mit 105 Tasten
  keyboard-configuration/variantcode:
  keyboard-configuration/unsupported_layout: true
  debian-installer/console-setup-udeb/title:
* keyboard-configuration/altgr: The default for the keyboard layout
* keyboard-configuration/ctrl_alt_bksp: false
  keyboard-configuration/layoutcode: de
  keyboard-configuration/other:
* keyboard-configuration/variant: Deutsch
```

`/home/$user/.config/wayfire.ini` 
```
...
[input]
xkb_model = pc105
xkb_layout = gb
xkb_variant =
```

CvH, 2024-09-23:
static workaround that does the job 👍 

`pi-gen/stage4/07-workaround/00-run.sh`
```
#!/bin/bash -e

sed -E -i \
  -e "s/^xkb_model.*/xkb_model = pc105/" \
  -e "s/^xkb_layout.*/xkb_layout = de/" \
  -e "s/^xkb_variant.*/xkb_variant = nodeadkeys/" \
  ${ROOTFS_DIR}/etc/wayfire/template.ini \
  ${ROOTFS_DIR}/etc/wayfire/gtemplate.ini
```

CvH, 2025-02-10:
static workaround since the labwc switch

`pi-gen/stage4/07-workaround/00-run.sh`
```
#!/bin/bash -e

sed --debug -E -i \
  -e "s/^XKB_DEFAULT_LAYOUT.*/XKB_DEFAULT_LAYOUT=de/" \
  ${ROOTFS_DIR}/etc/xdg/labwc/environment
echo 'XKB_DEFAULT_VARIANT=nodeadkeys' >>${ROOTFS_DIR}/etc/xdg/labwc/environment
```

XECDesign, 2026-02-17:
> Do you know by chance if the same issue could affect general locale settings issues? I try to set locale settings like timezone as well but when booting pi from the image it still uses all english locale and i can't set it in the config menu, the changes there are not getting applied to the system.

labwc ignores the usual system settings and wants the keyboard layout defined in its own config file instead. This shouldn't affect the timezone or locale settings.

However, this isn't an area of pi-gen that gets tested, so it's quite possible there are other issues which lead to breakages you're seeing. If you open a new issue with the config file to reproduce the problem and how you've determined that the settings are not being applied, I can take a look.