RPi-Distro/pi-gen #796: Multiple builds which only differ in a later stage to cache erlier stages

MexHigh, 2024-09-27:
I've added a stage "stage2a" which adds application specific things to the RPi OS Lite image (created by stage2). I want to create mutiple builds (optimally from a build pipeline) which only differ in the stage2a stage, meaning that everything up to stage2 remains unchanged.

Every stage starting from stage3 has `SKIP` and `SKIP_IMAGE` files in it so that the build process stops after stage2a. This works good.

However, when running multiple builds sequencially, stage1 and stage2 are never "cached" and rebuilt, which takes a lot of time. So my question is: **Is there a way to cache build stages up to a specific stage to speed up multiple builds?**. I'm using the docker build method (`./build-docker.sh`).

MexHigh, 2024-10-09:
But wouldn't that mean that a stage n+1 would include all changes made in stage n? I want to make e.g. five images, all with "unique" settings starting from stage 2 (Lite image).

SithRoboticist, 2025-04-29:
@MexHigh did you resolve this eventually? If so, can you please share how?

MexHigh, 2025-04-29:
@SithRoboticist Sadly no. I stuck with my described approach for a while, but then took the completely different approach of simply taking the prebuilt images and customizing them.