RPi-Distro/pi-gen #799: Failed to build image with minimum change

chuongle, 2024-10-10:
I don't know what's wrong. I pretty much follow the instruction
- Git clone the repo
- Add SKIP to stage 3 4 5 because I want to build a lite image
- Add a config like this
```
IMG_NAME="MyImage"
RELEASE="bookworm"
TIMEZONE_DEFAULT="America/Chicago"
ENABLE_SSH="1"
PUBKEY_SSH_FIRST_USER=""
PUBKEY_ONLY_SSH="1"
```
- Run ./build.sh

I have tried this on both Raspberry PI 3b+ and Ubuntu laptop (with `master` branch and `2024-07-04-raspios-bookworm` tag) . I always encountered these errors:

```W: Retrying failed download of http://raspbian.raspberrypi.com/raspbian/pool/main/p/pinentry/pinentry-curses_1.2.1-1_armhf.deb
I: Retrieving pinentry-curses 1.2.1-1
I: Validating pinentry-curses 1.2.1-1
W: Retrying failed download of http://raspbian.raspberrypi.com/raspbian/pool/main/p/pinentry/pinentry-curses_1.2.1-1_armhf.deb
I: Retrieving pinentry-curses 1.2.1-1
I: Validating pinentry-curses 1.2.1-1
W: Retrying failed download of http://raspbian.raspberrypi.com/raspbian/pool/main/p/pinentry/pinentry-curses_1.2.1-1_armhf.deb
I: Retrieving pinentry-curses 1.2.1-1
I: Validating pinentry-curses 1.2.1-1
W: Retrying failed download of http://raspbian.raspberrypi.com/raspbian/pool/main/p/pinentry/pinentry-curses_1.2.1-1_armhf.deb
I: Retrieving pinentry-curses 1.2.1-1
I: Validating pinentry-curses 1.2.1-1
W: Retrying failed download of http://raspbian.raspberrypi.com/raspbian/pool/main/p/pinentry/pinentry-curses_1.2.1-1_armhf.deb
I: Retrieving pinentry-curses 1.2.1-1
I: Validating pinentry-curses 1.2.1-1
W: Retrying failed download of http://raspbian.raspberrypi.com/raspbian/pool/main/p/pinentry/pinentry-curses_1.2.1-1_armhf.deb
I: Retrieving pinentry-curses 1.2.1-1
I: Validating pinentry-curses 1.2.1-1
W: Retrying failed download of http://raspbian.raspberrypi.com/raspbian/pool/main/p/pinentry/pinentry-curses_1.2.1-1_armhf.deb
I: Retrieving pinentry-curses 1.2.1-1
I: Validating pinentry-curses 1.2.1-1
W: Retrying failed download of http://raspbian.raspberrypi.com/raspbian/pool/main/p/pinentry/pinentry-curses_1.2.1-1_armhf.deb
I: Retrieving pinentry-curses 1.2.1-1
I: Validating pinentry-curses 1.2.1-1
W: Retrying failed download of http://raspbian.raspberrypi.com/raspbian/pool/main/p/pinentry/pinentry-curses_1.2.1-1_armhf.deb
I: Retrieving pinentry-curses 1.2.1-1
I: Validating pinentry-curses 1.2.1-1
W: Retrying failed download of http://raspbian.raspberrypi.com/raspbian/pool/main/p/pinentry/pinentry-curses_1.2.1-1_armhf.deb
W: http://raspbian.raspberrypi.com/raspbian/pool/main/p/pinentry/pinentry-curses_1.2.1-1_armhf.deb was corrupt
W: Couldn't download package pinentry-curses (ver 1.2.1-1 arch armhf) at http://raspbian.raspberrypi.com/raspbian/pool/main/p/pinentry/pinentry-curses_1.2.1-1_armhf.deb
I: Retrieving procps 2:4.0.2-3
I: Validating procps 2:4.0.2-3
I: Retrieving raspbian-archive-keyring 20120528.2
I: Validating raspbian-archive-keyring 20120528.2
W: Retrying failed download of http://raspbian.raspberrypi.com/raspbian/pool/main/r/raspbian-archive-keyring/raspbian-archive-keyring_20120528.2_all.deb
I: Retrieving raspbian-archive-keyring 20120528.2
I: Validating raspbian-archive-keyring 20120528.2
W: Retrying failed download of http://raspbian.raspberrypi.com/raspbian/pool/main/r/raspbian-archive-keyring/raspbian-archive-keyring_20120528.2_all.deb
I: Retrieving raspbian-archive-keyring 20120528.2
I: Validating raspbian-archive-keyring 20120528.2
W: Retrying failed download of http://raspbian.raspberrypi.com/raspbian/pool/main/r/raspbian-archive-keyring/raspbian-archive-keyring_20120528.2_all.deb
I: Retrieving raspbian-archive-keyring 20120528.2
I: Validating raspbian-archive-keyring 20120528.2
W: Retrying failed download of http://raspbian.raspberrypi.com/raspbian/pool/main/r/raspbian-archive-keyring/raspbian-archive-keyring_20120528.2_all.deb
I: Retrieving raspbian-archive-keyring 20120528.2
I: Validating raspbian-archive-keyring 20120528.2
W: Retrying failed download of http://raspbian.raspberrypi.com/raspbian/pool/main/r/raspbian-archive-keyring/raspbian-archive-keyring_20120528.2_all.deb
I: Retrieving raspbian-archive-keyring 20120528.2
I: Validating raspbian-archive-keyring 20120528.2
W: Retrying failed download of http://raspbian.raspberrypi.com/raspbian/pool/main/r/raspbian-archive-keyring/raspbian-archive-keyring_20120528.2_all.deb
I: Retrieving raspbian-archive-keyring 20120528.2
I: Validating raspbian-archive-keyring 20120528.2
W: Retrying failed download of http://raspbian.raspberrypi.com/raspbian/pool/main/r/raspbian-archive-keyring/raspbian-archive-keyring_20120528.2_all.deb
I: Retrieving raspbian-archive-keyring 20120528.2
I: Validating raspbian-archive-keyring 20120528.2
W: Retrying failed download of http://raspbian.raspberrypi.com/raspbian/pool/main/r/raspbian-archive-keyring/raspbian-archive-keyring_20120528.2_all.deb
.......
I: Validating udev 252.30-1~deb12u2+rpi1
I: Retrieving usr-is-merged 37~deb12u1
I: Validating usr-is-merged 37~deb12u1
I: Retrieving util-linux 2.38.1-5+deb12u1
I: Validating util-linux 2.38.1-5+deb12u1
I: Retrieving util-linux-extra 2.38.1-5+deb12u1
I: Validating util-linux-extra 2.38.1-5+deb12u1
I: Retrieving vim-common 2:9.0.1378-2
I: Validating vim-common 2:9.0.1378-2
I: Retrieving vim-tiny 2:9.0.1378-2
I: Validating vim-tiny 2:9.0.1378-2
I: Retrieving zlib1g 1:1.2.13.dfsg-1
I: Validating zlib1g 1:1.2.13.dfsg-1
E: Couldn't download packages: raspbian-archive-keyring dmsetup netbase libtirpc-common libuchardet0 libunistring2 libtext-wrapi18n-perl libassuan0 libtext-charwidth-perl paxctld init coreutils login base-passwd libxxhash0 libgpg-error0 libpopt0 libnettle8 gcc-8-base libgmp10 install-info libdevmapper1.02.1 libgcrypt20 passwd net-tools libnpth0 libjson-c5 libpipeline1 iputils-ping libjansson4 gzip mawk libedit2 pinentry-curses libmd0 libhogweed6 gcc-7-base dmidecode libtasn1-6 init-system-helpers libtirpc3 libmnl0 libreadline6 libnftnl11 libgdbm6 cpio libsasl2-modules-db libsigc++-1.2-5c2 netcat-openbsd gcc-9-base libbz2-1.0 libsasl2-2 libffi8 libsemanage-common libtext-iconv-perl
rmdir: failed to remove '/home/chirp/pi-gen/work/ChirpHubian/stage0/rootfs/debootstrap': Directory not empty
[13:34:09] bootstrap failed: please check /home/chirp/pi-gen/work/ChirpHubian/stage0/debootstrap.log
[13:34:09] Build failed
```

It basically can't download any packages. If I click on the link of those packages it will redirect me to `http://raspbian.freemirror.org/raspbian/pool/main/r/raspbian-archive-keyring/...` which is not available.

The only way for me to get it working is switch over to branch `arm64` and i can get the `build-docker.sh` running on my ubuntu.

FacuM, 2024-11-03:
Do you happen to have an RPi at hand? I was having all sorts of connectivity issues with the QEMU emulator and now I'm trying with an RPi 4, it's flying fast.

chuongle, 2024-11-04:
I have not tried on a Pi4 yet. I have tried on a Pi3 with the same issue (at the time of creating this issue). But I just tried this again recently and it worked with the master branch. Maybe the domain has been fixed for these package?

XECDesign, 2024-11-04:
It looks like a network issue. debootstrap couldn't download some files, so there's no much that can be done on pi-gen side. You may be behind a bad proxy or maybe you were hitting a bad Raspbian mirror.

closed 2024-11-04 by XECDesign